ebroecker/canmatrix #130: Including PGN and SPNs in xlsx exports

saleenman95, 2018-01-15:
Would it be possible to include the PGN and SPN (for J1939 DBCs) in xlsx exports?  This would make it nice for using VLOOKUP while making a list of PGN and SPNs for a project.  Since one could just specify the SPN and get the Units, Scaling, and Offset to automatically populate.

Thanks!

ebroecker, 2018-03-22:
I'm a dummy regarding j1939 
I found some information about PGN ((id >> 8) & int('0xFFFF', 16))
But what is the "SPN", maybe you could give me some lesson :-)

milhead2, 2018-03-22:
SPN's are what J1939 uses to label specific fields within a PGN.   Some a documented in the SAE standards but most companies add their own.

For example: 
    A J1939 PGN to report generic buttons 'pOperator Switch Status' is defined as PGN 65381.  The first field in the PGN is a 2 bit field called 'pOperator Button 1' and is defined as SPN 524059..  There can be a rash of other SPN's in the PGN..

I don't know the full context of saleenman95's quesiton but when I used canmatrix to convert from a Vector format to json I got all the SPN definitions within each PGN

Mil

saleenman95, 2018-05-16:
Oh wow, I need to check on this stuff more frequently... Sorry for the delay guys.

@ebroecker, @milhead2 got it perfectly.  SPNs aren't really encoded in anything, they are just stored in the Attributes section for a given signal.  

@milhead2 That PGN and SPN combo looks awefully familiar... ;)  I think we're working towards the same goal here, I just wanted it in the XLSX format, but if it exports in JSON, then I might be okay.

@altendky A J1939 DBC just has the CAN Arbitration ID (just like a standard DBC), the only difference is that CANdb++ recognizes the DBC format defined up top, and then does the decoding to determine the PGN, Source Address, Destination Address, and Priority.

ebroecker, 2018-05-17:
@saleenman95 

Thanks for clearing thinks so that even I understand it now.

You could archive this for now with following command-line (not very comfortable but working):

```bash
canconvert --additionalFrameAttributes=pgn --additionalSignalAttributes=attributes[\"SPN\"] source.dbc target.xlsx
```

I want to cleanup the attributes anyway. With this I'll add a property to the Signal, and afterwards it should be possible to just write `--additionalSignalAttributes=spn`

ebroecker, 2018-05-18:
not it should be possible:

```bash
canconvert --additionalFrameAttributes=pgn --additionalSignalAttributes=spn source.dbc target.xlsx
```

commit e8e24158 (2018-05-18): add spn as signal-attribute fixes #130
closed 2018-05-18 by ebroecker