ebroecker/canmatrix #132: Decode/Encode chinese

ShiboJiang, 2018-01-25:
Hi e:
    Thanks a lot for the god tool(神器)!
    I meet a issue when converting dbc to csv.  DBC file has chinese but the csv file display mess code. I try to convert dbc file to different encoding formate (GBK/utf-8), this issue has always exited.

ebroecker, 2018-01-25:
You need to tell canmatrix how dbc is encoded.
Even comments can have other coding than rest of dbc.

maybe something like following helps:
`./convert.py --dbcCommentCharset=GB2312 --dbcCharset=gbk input.dbc output.csv `

replace _GB2312_ and _gbk_ with your actual expected charsets in dbc

maybe you could see some infos about chinese charsets in (#3)
Please give feedback, if this solves your issue

ShiboJiang, 2018-01-29:
Use this way. I try the command `python convert.py --dbcImportEncoding=gbk --dbcImportCommentEncoding=gbk test.dbc test.xls` in canmatrix ver. 0.6 , and solve the chinese encoding issue. Thanks a lot.
By the way ,the factor and offset are not displayed clearly, and i find there is a old issue closed about that. You suggest json is better than excel,  maybe excel is a more effictively file for non-tech people emmm....
I will DIY this funtion, and also hope it come true in future version.
Thanks!

closed 2018-01-29 by ShiboJiang

ebroecker, 2018-01-29:
There is a 'bleeding edge' feature, maybe it works already:

`python convert.py --dbcImportEncoding=gbk --dbcImportCommentEncoding=gbk --additionalSignalAttributes signal.min,signal.max test.dbc test.xls`