ebroecker/canmatrix #136: JSON conversion - add values

gesc31, 2018-02-16:
Hi,
I used canconvert to convert a dbc file to a json file with jsonAll option but noticed that the values where not exported. 
Could you please modify the dump function in cmjson.py to add "values": signal.values, in signalDict by default?
Thanks

ebroecker, 2018-04-06:
Add even more information to --jsonAll
Goal should be to add all information, so that writing to json and reading again is loseless.

Seneda, 2020-04-24:
I have made a fix that add the Value Tables/Enumerations Dict to the JSON output file in [Seneda/canmatrix/tree/AddEnumstoJSON](https://github.com/Seneda/canmatrix/tree/AddEnumstoJSON). But I will hold off on doing a PR until i see if my existsing PR #475 gets accepted or not so that I can rebase my branch before submitting it.

ebroecker, 2022-05-27:
can this issue be closed?

ebroecker, 2023-01-25:
https://github.com/ebroecker/canmatrix/commit/32fb4932b6d9124932ca8fed7c0b2c7caea6edf0

closed 2023-01-25 by ebroecker