ebroecker/canmatrix #138: How to get Offset field in Xls file after DBC to Xls conversion

Kashambabu, 2018-02-26:
Hi,

I am converting DBC file to Xls file but i am unable to get the Offset field in xls file,
Please help me in this one

Best regards,
Kasham Babu

ebroecker, 2018-02-26:
there is a "signal-byte" and a "signal-bit" field in the xls. 
Whats wrong with them? 

Please describe your issue a bit more in detail, maybe with examples, this would help us to help you...

ebroecker, 2018-03-22:
since here is no answer - seems this issue can be closed.
Please reopen, if issue is still there

closed 2018-03-22 by ebroecker

zoumi, 2023-05-20:
@ebroecker the offset field is missing:
For example dbc:
```
BO_ 784 Car_Temp_info_0: 8 Vector__XXX
 SG_ car_T_01_00 : 4|13@0+ (0.1,-40) [-40|125] "dec" Vector__XXX
```
convert it to xls then convert it back to dbc, you will get:
```
BO_ 784 Car_Temp_info_0: 8 Vector__XXX
 SG_ car_T_01_00 : 4|13@0+ (0.1,0) [0|8191] "dec" Vector__XXX
```
the offset value "-40" is missing.

ALUE, 2023-05-20:
您的邮件已收到！              祝好             刘毅