ebroecker/canmatrix #139: File dumped as sym file (PEAK) doesn't open in PCAN Symbol Editor - wrong ID format

david-dego, 2018-02-27:
Using the library directly as API and dumping a canmatrix object as a sym file,  gives a file where the ID has an extended format even though the type is normal

The consequence is that it's not recongnized by PEAK's software 'PCAN Symbol Editor' unless the specified type is "extended", the software gives an error when trying to open the file. It works OK when removinf the extra zeros on the left.

Example : 
```
ID=00000050h 
# doesn't open in PCAN SE
```

```
ID=00000050h 
Type=Extended
#opens  well in PCAN SE
```

```
ID=050h
#opens in PCAN SE
```

I think it's on PCAN SE's side that is should be able to accepts an ID with the extended format even though it's in the 11bits type. But it would help to conform to their format.

david-dego, 2018-02-27:
#140 Created a pull request for this

ebroecker, 2018-02-28:
fixed with pr #140

closed 2018-02-28 by ebroecker