ebroecker/canmatrix #144: handling of extended CAN id

danielhrisca, 2018-04-04:
Hello,

I have a DBC file that contains the following definition:

```
BO_ 2222222222 MSG_4746B8E: 6 MSG
```

If is load this DBC and try to get the message using the extended ID, the frame is not found.

Should the user handle the extended ids?

ebroecker, 2018-04-04:
Hi,

not shure what you mean:
The arbitration ID is "0x4746b8e" (decimal: 74738574)

dbc does not have an attribute for "extended id" but uses bit 31 (start from 0) for marking the id as extended. Such id 2222222222 in dbc is  "0x4746b8e" (decimal: 74738574)

danielhrisca, 2018-04-04:
Hello Eduard,

I'm not sure if you are familiar with the MDF standard:
* in case of logging of a CAN bus there will be a channel group for each CAN message that will hold the metadata for the message 
* there is a linked database file that is used to parse the message payload
* the message id is referred in the channel group's name and XML comment

[image]
[image]

As you can see the id is given as extended value.

Do you think it makes sense to check if the Frame is extended here https://github.com/ebroecker/canmatrix/blob/development/canmatrix/canmatrix.py#L83 and also check if ``Id-0x80000000`` matches ``frame.id``?

ebroecker, 2018-04-04:
the bit 31 (vector, dbc, mdf) is the same as frame.extended (canmatrix)

you can do something like:
```
if frame.extended == True:
   frame.id += 0x80000000
```
or the way arround:

```
if frame.id > 0x80000000:
            frame.id -= 0x80000000
            frame.extended = True
```

you are right, `byId(self, Id):` does not distinguish between extended and normal id.
I could add some handling for it. But it would look like:
```
    def byId(self, Id, extended=None):
        """
        returns a Frame-Object by given Frame-ID
        """
        for test in self._list:
            if test.id == int(Id):
                if extended is not None:
                     if test.extended == extended
                          return test
                else:
                     return test
return None
```

danielhrisca, 2018-04-04:
I've done something similar https://github.com/danielhrisca/asammdf/blob/development/asammdf/mdf_v4.py#L457

but I was wondering if this could be included to be available to all _canmatrix_ users. 

Is there some other impact if you implement something like:

```python
def byId(self, Id):
    """
    returns a Frame-Object by given Frame-ID
    """
    Id = int(Id)
    extended = 0x80000000
    for test in self._list:
        if test.id == Id:
            return test
        elif test.extended and Id & extended:
            if test.id == Id - extended:
                return test
return None
```
?

 With this the user doesn't have to do extra checks and you also don't get false positives if the extended message doesn't exist but a normal message without the extended flag exists in the database.

ebroecker, 2018-04-04:
what about merging both approaches?

```python
def byId(self, Id, extended=None):
    """
    returns a Frame-Object by given Frame-ID
    """
    Id = int(Id)
    extendedMarker = 0x80000000
    for test in self._list:
        if test.id == Id and extended is None:
            # found ID while ignoring extended or standard
            return test
        elif extended is not None and test.extended == extended:
            # found ID while checking extended or standard
            return test
        elif test.extended and Id & extended:
            # check regarding common used extended Bit 31
            if test.id == Id - extendedMarker:
                return test
return None
```

danielhrisca, 2018-04-04:
I think you should use _entendedMarker_ instead of _extend_ for the masking

```python
def byId(self, Id, extended=None):
    """
    returns a Frame-Object by given Frame-ID
    """
    Id = int(Id)
    extendedMarker = 0x80000000
    for test in self._list:
        if test.id == Id and extended is None:
            # found ID while ignoring extended or standard
            return test
        elif extended is not None and test.extended == extended:
            # found ID while checking extended or standard
            return test
        elif test.extended and Id & extendedMarker:
            # check regarding common used extended Bit 31
            if test.id == Id - extendedMarker:
                return test
return None
```

Thank you!

danielhrisca, 2018-04-04:
I just had a second look and there are some bugs in the implementation: if the Id is not equal and you have matching extend flags then you return the current Frame 

Please have a look over this new proposal:
```python
def byId(self, Id, extended=None):
    """
    returns a Frame-Object by given Frame-ID
    """
    Id = int(Id)
    extendedMarker = 0x80000000
    for test in self._list:
        if test.id == Id:
            if extended is None:
                # found ID while ignoring extended or standard
                return test
            elif test.extended == extended:
                # found ID while checking extended or standard
                return test
        else:
            if extended is not None:
                # what to do if Id is not equal and extended is also provided ???
                pass
            else:
                if test.extended and Id & extendedMarker:
                    # check regarding common used extended Bit 31
                    if test.id == Id - extendedMarker:
                        return test
    return None
```

ebroecker, 2018-04-04:
hopefully this works...

altendky, 2018-04-04:
This belongs in an MDF parser.  If a program has a user entering MDF-formatted identifiers it should handle parsing them, just as it should handle converting the user-entered string to an integer.  canmatrix already translates this when reading DBC, correct?  CAN arbitration IDs simply do not have a 31st bit in any case, despite some file formats storing bit-packed data.  Perhaps it would help for canmatrix to provide explicit and separate `idFrom31BitId()` and `extendedFrom31BitId()` functions or somesuch.

Also, in the images each instance of 2222222222 looks weird...  they are different font sizes and seem to be affecting text around them.  Were the images edited?

This function should be very easy to both read and write.

```python
def byId(self, Id, extended=None):
    """
    returns a Frame-Object by given Frame-ID
    """
    if extended is not None:
        extended = bool(extended)

    for test in self._list:
        if extended is None:
            if test.id == Id:
                return test
        elif test.extended == extended:
            return test
```
And rather than returning `None` an exception would be nice but such a change should be considered separately.
```python
    if extended is None:
        extended_note = 'either an extended or standard ID'
    elif extended:
        extended_note = 'an extended ID'
    else:
        extended_note = 'a standard ID'
        
    raise SomeNotFoundException('No frame found with ID {:X} as {}'.format(extended_note))
```

danielhrisca, 2018-04-04:
@altendky 
I have edited the screenshot to hide the real values, but what I have said is valid: the message id value is the extended one.

altendky, 2018-04-04:
I suppose it could be argued that canmatrix is missing an arbitration id class.  I'm not sure if verifying the range of the id passed to the constructor would be good or not.

```python
class ArbitrationId:
    id_mask = (1 << 29) - 1
    extended_mask = 1 << 31

    def __init__(self, id, extended=None):
        self.id = int(id)

        if extended is None:
            self.extended = None
        else:
            self.extended = bool(extended)

    @classmethod
    def from31BitId(cls, id):
        return cls(
            id=id & self.id_mask,
            extended=id & self.extended_mask,
        )

    def __eq__(self, other):
        return (
            self.id == other.id
            and (
                self.extended is None
                or other.extended is None
                or bool(self.extended) == bool(other.extended)
            )
        )
```

I think I had a bug in my previous implementation of `byId()` but anyways, it's even simpler with this.

```python
def byId(self, Id):
    """
    returns a Frame-Object by given Frame-ID
    """
    for test in self._list:
        if Id == test:
            return test
```

danielhrisca, 2018-04-04:
@altendky 
Well you are back to the initial implementation

altendky, 2018-04-04:
@danielhrisca except it is conditionally sensitive to the extendedness.  Also, my overall point is that your issue of having compound id/extendedness integers should not be addressed inside the search function.  It should be addressed at the boundary where you receive this compound integer containing both the id and the extendedness.  If you are parsing an MDF you should convert there and if you are taking user input you should convert there.  Otherwise, it'd have to be reimplemented in each place that considers ID equality.  This becomes an inconsistent mess quickly.

danielhrisca, 2018-04-04:
As I've already stated, I've done the handling in the MDF parser, but the intention was to abstract this away from the users so that others can benefit from this. 

If the impact is unacceptable then keep the old implementation.

altendky, 2018-04-04:
It's certainly @ebroecker's call overall.

Do you not like the idea of having the parsing of the compound integer as a distinct feature?  I think it's perfectly reasonable for canmatrix to provide that.  It's probably already coded repetitively in a couple places I'd guess, which wouldn't be great.  I just don't think it's good to encourage people to have several formats for the same data spread throughout their program.  Similar to thinking this shouldn't be added to `byId()`, I also don't think `byId()` should parse hexadecimal strings like `'0x123'`, etc.  It's much easier if we work towards a single internal representation of any given data and handle conversions at the boundaries (user input, file parsing, etc).

It'd be a bit of work, which I am not presently volunteering to do so...  but my feeling is that the arbitration ID class is the proper solution and should replace the existing separate id and extended attributes.  It provides the parsing of MDF/DBC/etc-formatted extended identifiers so others don't have to reimplement it and also provides the equality testing in one place.  Raising an exception in `__init__()` for an extended compound ID having been passed would encourage people to learn to use the explicit class method for that. (detecting a standard compound id isn't really possible)

```python
class ArbitrationId:
    standard_id_mask = (1 << 11) - 1
    extended_id_mask = (1 << 29) - 1
    compound_extended_mask = 1 << 31

    def __init__(self, id, extended=None):
        self.id = id

        if extended is None:
            self.extended = None
        else:
            self.extended = extended

        if self.extended is None or self.extended:
            mask = self.extended_id_mask
        else:
            mask = self.standard_id_mask
        
        if self.id != self.id & mask:
            raise SomeError('ID out of range')

            
    @classmethod
    def fromCompoundInteger(cls, i):
        return cls(
            id=i & self.extended_id_mask,
            extended=i & self.compound_extended_mask,
        )

    def __eq__(self, other):
        return (
            self.id == other.id
            and (
                self.extended is None
                or other.extended is None
                or self.extended == other.extended
            )
        )
```

danielhrisca, 2018-04-04:
I'm not familiar with the code base so I can't have a strong opinion on the implications of this change. You guys know better what the impact is. My first impression was that the proposed change would be minor.

altendky, 2018-04-04:
@danielhrisca in many senses it is minor.  But, it encourages continued mislocation of features both in this project and in the projects of people using this one.

It's my impression that my solution would be more generally useful and would not only help people deal with searching but would help them deal better with anything related to arbitration id's.  Do you disagree?  I certainly may be missing a scenario that would be good to address.  Regardless it should have a `to_compound_integer()` as well.

<details>
  <summary>Yet another version of `ArbitrationId`</summary>

```python
class ArbitrationId:
    compounded_extended_bit = 31
    standard_id_mask = (1 << 11) - 1
    extended_id_mask = (1 << 29) - 1
    compound_extended_mask = 1 << compounded_extended_bit

    def __init__(self, id, extended=None):
        self.id = id

        if extended is None:
            self.extended = None
        else:
            self.extended = extended

        self.validate()

    def validate(self):
        if self.id != self.id & self.mask():
            raise SomeError('ID out of range')
            
    def mask(self):
        if self.extended is None or self.extended:
            return self.extended_id_mask
        
        return self.standard_id_mask
        
    @classmethod
    def from_compound_integer(cls, i):
        return cls(
            id=i & self.extended_id_mask,
            extended=i & self.compound_extended_mask,
        )

    def __eq__(self, other):
        return (
            self.id == other.id
            and (
                self.extended is None
                or other.extended is None
                or self.extended == other.extended
            )
        )

    def to_compound_integer(self):
        self.validate()

        return self.id & ((1 if self.extended else 0) << self.compounded_extended_bit)
```
</details>

ebroecker, 2018-04-06:
@altendky 
I agree, the ArbitrationId-Class would make things cleaner.
If I find some time, I'll integrate it in the codebase and look, what problems will come up :-).
Thanks again for these architecture hints.

ebroecker, 2019-02-26:
@altendky 
I added your Arbitration_Id class - maybe you find the time to have a quick look on it? 
PR #296

PR #296 (2019-02-26): [WIP] introduction of Arbitration_id class
should fix  #144
closed 2019-02-26 by ebroecker