ebroecker/canmatrix #146: Attributes

MichaelRSG, 2018-04-05:
Hi,

is it possible to add some attributes to the signals like cycle time in the Interaction Layer?

Thank you :)

altendky, 2018-04-05:
I'll note that this is available for sym format, but not as a general canmatrix feature yet I don't believe.

ebroecker, 2018-04-05:
currently the cycle time and the send type are in frame and signal attributes (frame.attributes and signal.attributes) only.

I want these both as direct properties (signal.cycleTime, signal.sendType, frame.cycleTime, frame.sendType).

ebroecker, 2018-04-06:
Physically (on CAN) a cycletime does only make sense on frames.
But if you move signals from one frame to another it is interesting to know the cycletime needs from the signal. 
In dbc it is possible to define cycletimes for signals. 
Frames can inherit the cycletimes from the signals (and use the fastest one) automatically.

I'll implement some frame properties which optionally inherit the cycletimes from the signals.
Hope this makes this issue a bit more clear.

closed 2019-11-08 by ebroecker