ebroecker/canmatrix #149: Question about convert the .arxml file to the .dbc file

xh20, 2018-05-16:
when i tried to convert an .arxml file to a .dbc file, i found this message: "ERROR - formate -This file format is not supported for reading", and i checked the command "canconvert" and found this command support 
cmcsv      for     export
dbc   for  import and  export
cmjson for  import  and export
sym    for import  and export
dbf    for import and  export
could someone help me to convert arxml file to dbc file??

ebroecker, 2018-05-17:
did you install lxml package for python?

xh20, 2018-05-17:
no, i just installed canmatrix, after i installed lxml package, the command support now more files, but expert arxml. Maybe i need to install more packages. Which other packages do i need to install?

ebroecker, 2018-08-17:
usually `pip install git+https://github.com/ebroecker/canmatrix#egg=canmatrix[arxml]` should be enough. lxml is the only special need for arxml.
Could you provide more information whats going wrong?

altendky, 2018-08-17:
I'd suggest a full console log including installing canmatrix, any other installation, a pip freeze, and the conversion and output.  And if possible the input file.  Preferably this is all done in a virtualenv or venv (that's how it should be done anyways).

If you share a particular os and python version I can provide (fairly) specific commands to run.

ebroecker, 2018-10-29:
nothing heared again

closed 2018-10-29 by ebroecker