ebroecker/canmatrix #152: OverflowError occurs when I try to arxml to dbc

lucyk4t, 2018-05-24:
`Traceback (most recent call last):
  File "d:\python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\Python36\Scripts\canconvert.exe\__main__.py", line 9, in <module>
  File "d:\python36\lib\site-packages\canmatrix\convert.py", line 294, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "d:\python36\lib\site-packages\canmatrix\convert.py", line 41, in convert

    dbs = canmatrix.formats.loadp(infile, **options)
  File "d:\python36\lib\site-packages\canmatrix\formats.py", line 54, in loadp
    return load(fileObject, importType, **options)
  File "d:\python36\lib\site-packages\canmatrix\formats.py", line 64, in load
    dbs = moduleInstance.load(fileObject, **options)
  File "d:\python36\lib\site-packages\canmatrix\arxml.py", line 1553, in load
    ns))
  File "d:\python36\lib\site-packages\canmatrix\arxml.py", line 1315, in getFram
e
    getSignals(signaltopdumaps, newFrame, arDict, ns, None)
  File "d:\python36\lib\site-packages\canmatrix\arxml.py", line 1112, in getSign
als
    is_float=is_float)
  File "d:\python36\lib\site-packages\canmatrix\canmatrix.py", line 240, in __in
it__
    self.setMax()
  File "d:\python36\lib\site-packages\canmatrix\canmatrix.py", line 433, in setM
ax
    self._max = self.calcMax()
  File "d:\python36\lib\site-packages\canmatrix\canmatrix.py", line 440, in calc
Max
    return self._offset + (rawMax * self._factor)
OverflowError: int too large to convert to float
`

I got issue for converting from arxml to dbc.
Could you check about this issue? Thank you.

altendky, 2018-05-29:
Are you running this from the CLI?  I'm not used to seeing `runpy` in the traceback so I might be reading this wrong.  This should be covered by the `float_factory` feature of `Signal` though I presume we don't have this accessible from the CLI.

lucyk4t, 2018-06-03:
Dear,
You're right. I just used CLI. Should I use IDE or other tools?

ebroecker, 2018-06-04:
@altendky 
Should the CLI use `Decimal` as `float_factory` for default?

ebroecker, 2018-08-11:
as I switched to Decimal now, this issue should be fixed

closed 2018-08-11 by ebroecker