ebroecker/canmatrix #156: Manufacturer Defined PGNs for J1939

rpkvt, 2018-06-13:
Hello!

I believe there's a need (at least I have a need) for a way to create dbc files which describe custom J1939 packets. J1939 has a set of PGNs  and SPN which describe the data in the payloads. J1939 also has a range of PGNs (e.g. 0xff00 through 0xffff) which are generic an are meant for a manufacturer to define. 

Is it possible to implement adding this information to canmatrix? An abstracted class like add J1939 signal would be awesome.   Is there a way to do this now?

Thanks,
Rich

ebroecker, 2018-06-13:
Well - I am willing to help. But not sure what really is needed. So my J1939-Knowledge is very basic.
You want something like "addJ1939Signal" which should do exactly what?
Maybe you could add some function-signature or some pseudo-code to make more clear what you would expect. 

Thanks

rpkvt, 2018-06-14:
Great, thanks for entertaining the idea! 

I have to admit that I know next to nothing about dbc files and I only know enough to be dangerous with J1939, so I may not know enough to help you make this a reality. I'll describe what I'm looking for and you can let me know if you think it's feasible.

My understanding is that dbc can be used to define the meaning behind data within a CAN message or string of CAN messages to a receiver.  I'm proposing we add to your code an abstracted way to define a J1939 message which is made up of one or multiple CAN messages.  

First a little structure:

J1939 uses extended CAN messages:
[image]

J1939 combines the 29 bits between SOF and RTR to create the J1939 PDU (protocol data unit), which is broken up like so:
[image]

At a high level, it would be great if the user provided a priority, pgn, source address to a J1939 specific method.  Another method would then add channels of data to the payload.  Once done adding channels to the packet, the user would commit and the I think the object would then fill out the rest  (e.g. CRC). 

To start, I think the manufacturer defined PGNs are what you'd want to support since everything else is already rigorously defined and the end user likely has these definitions already.  Manufacturer defined broacast (not destined for anyone in particular) PGNs I've used fall in the range of 0xFF00 to 0xFFFF but there are others.

For payloads that are 8 bytes or less, J1939 just uses one packet and pads remaining payload bytes with 0xFFs, so that's a fairly simple case.   

For payloads higher than 8 bytes, J1939 use a packet transport protocol. I think this is where things get a little tricky.   the first byte of the payload is a sequence number and the rest are the actual payload.

I just received a response from Vector about creating dbc files. Hopefully I can come up with an example that makes the end goal a little clearer.  I'll follow up with some more details in a bit.

rpkvt, 2018-06-14:
Here's my very first stab at a dbc file (renamed to txt):
[Example_J1939.txt](https://github.com/ebroecker/canmatrix/files/2103583/Example_J1939.txt)

Big caveat: I haven't tested this file to see if it works. So take it with a grain of salt. Maybe your understanding dbc files can help verify it.

This is how I set it up. I declared several signals (channels 1 through 8) as floats. I then created 3 messages with various channels in them.
[image]

[image]

In setting up the message, I defined the PGN, Priority, and the Source address:
[image]

And added specific signals to each message:
[image]

rpkvt, 2018-06-14:
So all that being said. I think we need the following:
- a method to create a general signal with various data types (u8,u16,u32,float,double,byte array, etc...).
- a method to define a message (pgn, priority, source)
- a method to add a signal to a message
- a method to commit the message

ebroecker, 2018-06-14:
thanks for this nice J1939-abstract.

Fortunately not everything is new to me :-)

* we have already methods to define messages (not with pgn, priority and source but only arbitration id)
* there are methods to create signals and add them to messages
* 'committing' messages is probably supported also

All of these methods are for classic can not for j1939. But as I meanwhile have some j1939 knowledge also, I'll create some j1939 friendly methods (with you posts as specification).

payloads bigger than 8 bytes are not really a problem, because they are supported already (the definitions in dbc are straight forward.)

rpkvt, 2018-06-14:
Great! 

Yeah, I thought that might be the case. I'm such a noob with dbc files though that I didn't really know what I was looking at with your code.   A j1939 friendly method would be much appreciated.

One thing that I didn't touch which may be important is the J1939 name.  This is a hex value which describes a device and is used by the device to claim an address.  

Here is some python code I wrote to create the J1939 name. 
[J1939Name.txt](https://github.com/ebroecker/canmatrix/files/2103819/J1939Name.txt)

The values in the dbc that seem to correspond to the name are here:
BA_DEF_DEF_  "ECU" "";
BA_DEF_DEF_  "NmJ1939SystemInstance" 0;
BA_DEF_DEF_  "NmJ1939System" 0;
BA_DEF_DEF_  "NmJ1939ManufacturerCode" 0;
BA_DEF_DEF_  "NmJ1939IndustryGroup" 0;
BA_DEF_DEF_  "NmJ1939IdentityNumber" 0;
BA_DEF_DEF_  "NmJ1939FunctionInstance" 0;
BA_DEF_DEF_  "NmJ1939Function" 0;
BA_DEF_DEF_  "NmJ1939ECUInstance" 0;
BA_DEF_DEF_  "NmJ1939AAC" 0;
BA_DEF_DEF_  "NmStationAddress" 254;

rpkvt, 2018-06-15:
I may have jumped the gun a bit. It looks like there is J1939 specific code, particularly in these files:

join.py 
canmatrix.py has the object CanId.

Is there more complete documentation on how to generate dbc files and what signals are versus frames?  I'm unable to run the code as is documented on the readme using the latest version installed through PIP.

ebroecker, 2018-06-15:
@rpkvt 
i created a bit support which hopefully help you:
have a look at examples/createJ1939Dbc.py

rpkvt, 2018-06-18:
Thanks!!!   This absolutely great. Thank you.

I'll mess around with it some more. I'll let you know if I find anything notable.

ebroecker, 2018-06-18:
don't bother about "VFrameFormat" - I'll move this to canmatrix internal, such that it won't be necessary in the example anymore...

ebroecker, 2018-07-18:
@rpkvt 
any wishes open?
 otherwise I'll close this issue

closed 2018-07-19 by ebroecker