ebroecker/canmatrix #158: Error while converting a dbc file to any other format

aggarwalpiyush, 2018-06-15:
Error is:
AttributeError: 'NoneType' object has no attribute 'group'
[image]

aggarwalpiyush, 2018-06-28:
Hi,
I have figured out the issue. The code is unable to handle following: "[ms]". So, when I replaced "[ms]" with "ms" in all places, it worked perfectly.
Thanks for the guidance.

ebroecker, 2018-06-29:
Hi,
thanks for this info.
What letters did these lines start with? This would help to fix the Bug.
thanks in advance

ebroecker, 2018-08-11:
sorry - I can't reproduce. 
Due lack of information, I have to close this issue.
It could be reopended, as soon as there are more infos about this issue.

closed 2018-08-11 by ebroecker