ebroecker/canmatrix #161: Build broken with new PyYaml==4.1 (or 4.1b1)

altendky, 2018-06-27:
https://travis-ci.org/ebroecker/canmatrix/jobs/397500683

```python-traceback
5.98s$ source ./test.sh
Obtaining file:///home/travis/build/ebroecker/canmatrix
Requirement already satisfied: future in /home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages (from canmatrix==0.6.post58+gdd34a77)
Requirement already satisfied: lxml in /home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages (from canmatrix==0.6.post58+gdd34a77)
Requirement already satisfied: xlrd in /home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages (from canmatrix==0.6.post58+gdd34a77)
Requirement already satisfied: xlwt in /home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages (from canmatrix==0.6.post58+gdd34a77)
Requirement already satisfied: xlsxwriter in /home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages (from canmatrix==0.6.post58+gdd34a77)
Requirement already satisfied: pyyaml in /home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages (from canmatrix==0.6.post58+gdd34a77)
Installing collected packages: canmatrix
  Running setup.py develop for canmatrix
Successfully installed canmatrix
arxml -> ['csv', 'dbc', 'dbf', 'json', 'kcd', 'sym', 'xls', 'xlsx', 'xml', 'yaml']
Traceback (most recent call last):
  File "./test.py", line 85, in <module>
    canmatrix.convert.convert(in_file, out_file)
  File "/home/travis/build/ebroecker/canmatrix/canmatrix/convert.py", line 198, in convert
    canmatrix.formats.dumpp(outdbs, outfileName, **options)
  File "/home/travis/build/ebroecker/canmatrix/canmatrix/formats.py", line 113, in dumpp
    dump(db, fileObject, exportType, **options)
  File "/home/travis/build/ebroecker/canmatrix/canmatrix/formats.py", line 89, in dump
    moduleInstance.dump(canMatrixOrCluster, fileObject, **options)
  File "/home/travis/build/ebroecker/canmatrix/canmatrix/yaml.py", line 68, in dump
    f.write(yaml.dump(newdb).encode('utf8'))
  File "/home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages/yaml/__init__.py", line 217, in dump
    return dump_all([data], stream, Dumper=Dumper, **kwds)
  File "/home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages/yaml/__init__.py", line 196, in dump_all
    dumper.represent(data)
  File "/home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages/yaml/representer.py", line 26, in represent
    node = self.represent_data(data)
  File "/home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages/yaml/representer.py", line 57, in represent_data
    node = self.yaml_representers[None](self, data)
  File "/home/travis/virtualenv/python3.5.5/lib/python3.5/site-packages/yaml/representer.py", line 229, in represent_undefined
    raise RepresenterError("cannot represent an object", data)
yaml.representer.RepresenterError: ('cannot represent an object', <canmatrix.canmatrix.CanMatrix object at 0x7fc24d0a3ef0>)
```

Nothing jumps out to me as an obvious cause in the [announcements](https://github.com/yaml/pyyaml/blob/d3eb7daf88e29bc20ffd4cfb833da3c49c617625/announcement.msg).  Maybe the 'safe by default' aspect though?

altendky, 2018-06-27:
On `#pyyaml` it was suggested I start with https://github.com/yaml/pyyaml/pull/74 and https://github.com/yaml/pyyaml/issues/177 to start learning what might be wrong.

ebroecker, 2018-06-27:
is there anybody out there who uses the yaml-format?
Maybe we just should kick it out?

altendky, 2018-06-27:
I can take a little time.  I don't actively use yaml anywhere but I wonder if I should.

ebroecker, 2018-06-28:
My plan was to add all information to json, so there would be no need for yaml anymore ...

altendky, 2018-06-28:
There are various differences between the formats so I don't think simply having all data in JSON alleviates and potential use of YAML.  But...  I don't really care.  :]  I was just following up on my commit resulting in a broken build, even if totally unrelated to my commit.

ebroecker, 2020-09-20:
for me yaml works.
If not, create a new issue...

closed 2020-09-20 by ebroecker