ebroecker/canmatrix #162: Environment variables were not captured.

Jayanthdj, 2018-06-28:
When I tried to merge two .dbc files, Environment variables were not captured. Could you please help

ebroecker, 2018-06-28:
Enviroment Variables are not supported yet.
I'll check the needed efford and perhaps help you...

Jayanthdj, 2018-06-28:
It would be great if you help me to capture environment variables. Thank you so much for your response!

Jayanthdj, 2018-07-04:
Hi Edward, any updates on this?

ebroecker, 2018-07-18:
merging complete dbcs should merge enviroment-variables now.

Note: There is no check for double defined variables yet

Jayanthdj, 2018-07-20:
Hi Edward, Thank you so much for the implementation. This is very helpful but my actual need is to check and merge double defined variables as well. so Could you please help

ebroecker, 2018-07-20:
should be done now

Jayanthdj, 2018-07-24:
Hi Edward, I am getting this below error when I tried to merge it two dbc files
C:\LegacyApp\Python27\Scripts>canconvert-script.py --merge=second.dbc source.dbc target.dbc
merge complete: second.dbc
ERROR - canmatrix - ID Conflict, could not copy/merge frame CFT_DD_DVP3  2d7h CFT_DD_DVP3
ERROR - canmatrix - ID Conflict, could not copy/merge frame CFT_DD_DVP2  412h CFT_DD_DVP2
ERROR - canmatrix - ID Conflict, could not copy/merge frame CFT_DD_DVP1  26dh CFT_DD_DVP1
ERROR - canmatrix - Name Conflict, could not copy/merge EnvVar Env_STS_RKE
ERROR - canmatrix - Name Conflict, could not copy/merge EnvVar Env_WHL_RR_SPD

ebroecker, 2018-07-24:
... well this is exactly what you want, the ERRORS are the collisions:

seems Frame CFT_DD_DVP3, CFT_DD_DVP2  and CFT_DD_DVP1  are in both dbcs. So merge only takes the ones from "soruce.dbc"

Same with Enviroment-Variables Env_STS_RKE and Env_WHL_RR_SPD are in both dbcs also.
They are taken from source.dbc also.

ebroecker, 2018-07-26:
do you have any further  issues here or could I close this issue?

closed 2018-08-02 by ebroecker