ebroecker/canmatrix #165: Request Excel_template.xlsx for convert can_matrix_excel.xlsx to output.dbc

shaolinkhoa, 2018-07-05:
Hi,
Thank you for creating canmatrix. It's a great tool.

I converted the **input.dbc** to the **output.xlsx**. 
In the **output.xlsx**, there are top border lines and some cells have black background like the picture below. 
[image]

I have some questions about the convert processing. Would you mind helping me ?

1/ Does the top border lines contain message frame and all the signals of the frame ?
2/ what does the black background cell mean ?
3/ Does it have a standard excel template for converting a **excel_file.xlsx** to **output.dbc** ?

Thank you and regards,
Khoa

ebroecker, 2018-07-19:
1/ what is the exact question? The Top line should show collumn heads and all ECUs
2/ the different backgrounds are for better human readability only. No special Information
3/ you can create one by exporting on dbc to excel format. But excel is not fully describing a can matrix, there are informations, that only in other can be stored

shaolinkhoa, 2018-07-20:
Thank you for replying me.

Because i have to create dbc file from the information which are given in excel file. 
And create dbc file manually by CANdb++ are hard and time consuming so I need a tool can convert excel to dbc. Canmatrix is the best tool I found.
If you know any tools can create dbc file from excel file or any solutions, please share with me.

ebroecker, 2018-07-20:
Well - for starting - you can use every xls(x) which canmatrix creates.
I took the one which is created out of test.dbc and made all text visible again.
While most collumns should be self explainary, some hints:
* "Signal Function" is the comment for a Signal
* testBU and recBU are my example ECUs (Boardunits) 
* if there is one or more Values defined, they are used to create a Valuetable for this Signal.
* Name / Phys. Range either holds the value (if a value was defined in "Value" collumn) otherwise it holds the physical range (minValue..maxValue)
* "Function / Increment Unit" holds the Unit of the value (if a value was defined in "Value" collumn) otherwise it holds the increment of the physical value and seperated by space the Unit (this is than a Must)

[cmTemplate.xlsx](https://github.com/ebroecker/canmatrix/files/2212598/cmTemplate.xlsx)

shaolinkhoa, 2018-07-20:
Thank you.
I'm familiar with the excel template and I can convert xlsl to dbc now.
But I still wonder how to configure the offset value in excel file to convert to dbc ?
I see it automatically set offset equal minValue in dbc file after it's converted.

ebroecker, 2018-07-20:
Yes,

minValue..maxValue 

creates: min = minValue; offset = minValue; 

maybe I could add some collums to add offset and factor direct

ebroecker, 2018-07-20:
[cmTemplate.xlsx](https://github.com/ebroecker/canmatrix/files/2215640/cmTemplate.xlsx)

example for importing additional attributes

shaolinkhoa, 2018-07-23:
Thank you. I see you clear the "Vector__XXX" column.
But what do "Launch Parameter","Signal Default","Signal Not Available" & "frame.is_fd" do and prefer when convert xlsx to dbc ?

ebroecker, 2018-07-23:
Hi shaolinkhoa,

use or use not the collums, as you need. 
All collums afer "_Function / Increment Unit_" are free configurable. The collums in front of this collumn should not be changed.

**use frame.[ATTRIBUTE] to change the frame:**
ATTRIBUTE for Frame can be:
- size
- extended
- is_complex_multiplexed
- is_fd
- comment
- pgn
- priority

**use signal.[ATTRIBUTE] to change the signal:**
ATTRIBUTE for Frame can be:
- startBit
- signalsize
- is_little_endian
- is_signed
- factor
- offset
- min
- max
- unit
- comment
- multiplex
- mux_value
- is_float
- comments

**Note**
if value is defined more than once, the last collumn matters; Not really tested

ebroecker, 2018-07-26:
does this fulfill your requirements? Could I close this issue?

shaolinkhoa, 2018-07-27:
Yes. it's working perfect.
I will tell you if I find any problems. 
Thank you for supporting me.

shaolinkhoa, 2018-09-25:
Hi ebroecker,

I find a problem while converting xlsx -> dbc
I'm using signal.offset , signal.min, signal.max
Would you mind helping me.

1/ Can you add signal.init ? so that the converted dbc file will have initial value.
2/ If we have a value table of a signal like below.

```
Value   | Name / Phys. Range   | Function / Increment Unit | signal.offset  | signal.min   | signal.max
0         | 0x0000-0xFFFE:Valid   | 0.0030518  %                   | -100            | 10              | 100                
65535 | 0xFFFF:Invalid             | 0.0030518  %                   | -100            | 10              | 100
```

Note: if signal.offset (line 1) = -100 and signal.offset (line 2) = 0. 
The dbc will take signal.offset = -100 from line 1 and bypass signal.offset from line 2.

Problem with signal.max : After I converted xlsx -> dbc,  the maximum value of signal is 65535. It's not match with signal.max = 100 like I write above.
Would you mind checking it.

Thank you

shaolinkhoa, 2018-10-04:
Hi ebroecker,

Does the DLC of frame is fixed to DLC 8 ? 

I found in src/canmatrix/xlsx.py: 
```
            dlc = 8
            launchParam = getIfPossible(row, 'Launch Parameter')
            if type(launchParam).__name__ != "float":
                launchParam = 0.0
            launchParam = str(int(launchParam))

            if frameId.endswith("xh"):
                newBo = Frame(frameName, id=int(frameId[:-2], 16), size=dlc, extended=True)
            else:
newBo = Frame(frameName, id=int(frameId[:-1], 16), size=dlc)
```

I added the column frame.size and changed the value to 6 but it didn't work.
I think frame.size doesn't work if the frame is fixed to DLC 8 like code above. 
p/s: Do you have any news about the value table + max value ? 

Regards,

shaolinkhoa, 2018-10-17:
Hi ebroecker,
Do you have any update version for those situations above ?
Regards,

ebroecker, 2018-10-29:
not sure if it was actively fixed, but my quick test worked:
I added a collumn: frame.size and set "7" in the row of the frame.
Created dbc is ok.

NB: there is a dlc-check which raises the dlc, if the given dlc is to small for handling all signals.

closed 2018-10-29 by ebroecker

shamakh96, 2020-03-26:
i successfully run this command and the .xlsx is generated 
`canconvert  myDbc.dbc  myExcelA.xlsx `

i tried to run the following command :
`canconvert  myExcelA.xlsx myDbc.dbc `
but it prints the this error :  **"  No module named 'xlwt'  "** 

Am i doing it wrong ? :'D

ebroecker, 2020-03-26:
https://canmatrix.readthedocs.io/en/latest/installation.html#dependencies

you need to install xlwt python package