ebroecker/canmatrix #168: update int(min) to int(float(min))

toolsh, 2018-07-11:
The 4 line codes need update from int(xxx) to int(float(xxx))

- Because `int("1.84467440737096E+019")` will raise 
`ValueError: invalid literal for int() with base 10: '1.84467440737096E+019'`
- While `int(float(1.84467440737096E+019))` can fix the ValueError.

https://github.com/ebroecker/canmatrix/blob/82ece227cd2b5a7ced2e0f75610708ff1231ca04/canmatrix/canmatrix.py#L972
https://github.com/ebroecker/canmatrix/blob/82ece227cd2b5a7ced2e0f75610708ff1231ca04/canmatrix/canmatrix.py#L973
https://github.com/ebroecker/canmatrix/blob/82ece227cd2b5a7ced2e0f75610708ff1231ca04/canmatrix/canmatrix.py#L995
https://github.com/ebroecker/canmatrix/blob/82ece227cd2b5a7ced2e0f75610708ff1231ca04/canmatrix/canmatrix.py#L996

altendky, 2018-07-11:
Why is an integer define being constructed from floating point values?

altendky, 2018-07-11:
Also, that'd actually result in the wrong value while `decimal.Decimal` can get it correct.

``` python
>>> s = '1.84467440737096E+019'
>>> int(float(s))
18446744073709600768
>>> int(decimal.Decimal(s))
18446744073709600000
```

altendky, 2018-07-12:
What is the scenario you have where you are loading that representation of an integer into the define?

toolsh, 2018-07-13:
@altendky 
just the .dbc has definition similar as below:

> BA_DEF_  "BaudRate" FLOAT 1000 1000000;
BA_DEF_ BO_  "GenMsgSendType" ENUM  "cyclic","spontaneous","cyclicIfActive","spontaneousWithDelay","immediate";
BA_DEF_ BO_  "GenMsgCycleTime" FLOAT 0 300000;
BA_DEF_ BO_  "GenMsgDelayTime" FLOAT 0 300000;
**BA_DEF_ SG_  "GenSigStartValue" INT -9.22337203685478E+018 1.84467440737096E+019;**
BA_DEF_ SG_  "GenSigLongName" STRING ;
BA_DEF_DEF_  "BaudRate" 1000;
BA_DEF_DEF_  "GenMsgSendType" "";
BA_DEF_DEF_  "GenMsgCycleTime" 0;
BA_DEF_DEF_  "GenMsgDelayTime" 0;
BA_DEF_DEF_  "GenSigStartValue" 0;
BA_DEF_DEF_  "GenSigLongName" "";

altendky, 2018-07-13:
I'm wondering if we should verify that there is no actual fractional portion.  This would be most similar to how it is now.  If not, should we truncate?  Round?

toolsh, 2018-07-13:
The principle should be to get dbc read as accurate as possbile, plus not get canmatrix stuck.

for this line:
> BA_DEF_ SG_ "GenSigStartValue" INT -9.22337203685478E+018 1.84467440737096E+019;

3 ways:
-  `int(max)`, the one currently in use, will cause exception when use canmatrix to load dbc
-  `int(float(max))`, will get max = 18446744073709600768 which is slightly different from 1.84467440737096E+019 from mathatic point of view.
- `int(decimal.Decimal(s))`, will get max = 18446744073709600000 which is the same as 1.84467440737096E+019 from mathatic point of view.

Apparently, the last one as @altendky suggest is the best.  Since dbc is the only input to canmatrix to load its content, it's reasonable to follow exactly the same as what has been defined in dbc file and try to parse it literally to continue the load process, but not "what make it show in scientific notation?", "is there any mistake that it shows in scientific notation?" or "shall we fix possbile errors and correct it?". On the other hand, there is no additional input, besides dbc file, can be used for verification or decision of adjustment.