ebroecker/canmatrix #170: Some messages are not captured in .dbc file when converted from .arxml. Could you please help

Jayanthdj, 2018-07-20:
Some messages are not captured in .dbc file when converted from .arxml like
PDU Length, PDU Position, Min Update Time, Supervise Timeout, MsgIDType and MsgType .
Could you please help.

ebroecker, 2018-07-20:
well - where should these informations set in the dbc?
These would be properitary attributes in the dbc?

do you have an example arxml for this and maybe an idea where to store in dbc?

Jayanthdj, 2018-07-23:
I think this image may give an idea

[image]

Jayanthdj, 2018-07-23:
I think this image may give an Idea
[image]

ebroecker, 2018-08-11:
hmm ... these are probably attributes in the dbc.
Do you have some example-dbc ?

ebroecker, 2018-08-17:
I have no idea how/what to do here. Even with candb++ in the hand, I could not find these parameters...

ebroecker, 2018-11-02:
sorry, more information needed

closed 2018-11-02 by ebroecker