ebroecker/canmatrix #171: Error for some of the dbc files converting to kcd

mkbo, 2018-07-24:
vc-complex-type.2.4.a: Invalid content was found starting with element 'Value'. One of '{"http://kayak.2codeornot2code.org/1.0":MuxGroup}' is expected.]

Reason:
1. Multiplexor is not in the sequence order as per KCD Schmea definition. 
      <xs:sequence>
            <xs:element ref="MuxGroup" maxOccurs="unbounded"/>
            <xs:element ref="Notes" minOccurs="0"/>
            <xs:element ref="Consumer" minOccurs="0"/>
            <xs:element ref="Value" minOccurs="0"/>
            <xs:element ref="LabelSet" minOccurs="0"/>
          </xs:sequence>

     https://github.com/dschanoeh/Kayak/blob/master/Kayak-kcd/src/main/resources/com/github/kayak/canio/kcd/loader/Definition.xsd

seems the below changes working fine: 
commented 
                    #multiplexor.append(value)
                   #multiplexor.append(labelset)
and added the above lines under: 
                    if empty == 1:
                        multiplexor.append(muxgroup)
                        multiplexor.append(value)
                        multiplexor.append(labelset)
                message.append(multiplexor)

ebroecker, 2018-07-26:
is this fixed with #174?

ebroecker, 2018-08-16:
I think the PR has fixed this. If not - please reopen.

closed 2018-08-16 by ebroecker