ebroecker/canmatrix #172: Error while converting to KCD's if there is a producer:

mkbo, 2018-07-24:
vc-complex-type.2.4.a: Invalid content was found starting with element

     <xs:sequence>
        <xs:element minOccurs="0" ref="Notes"/>
        <xs:element minOccurs="0" ref="Producer"/>
        <xs:element minOccurs="0" ref="Multiplex" maxOccurs="unbounded"/>
        <xs:element maxOccurs="unbounded" ref="Signal" minOccurs="0"/>
      </xs:sequence>

Fix: The code seems not as per schema defined: 
 https://github.com/dschanoeh/Kayak/blob/master/Kayak-kcd/src/main/resources/com/github/kayak/canio/kcd/loader/Definition.xsd

The below code changes seems working fine: 

moved the notes to above producer:
           #comment = etree.Element('Notes')
            #if frame.comment is not None:
            #    comment.text = frame.comment
            #    message.append(comment)

moved the signal to below multiplex:
           # standard-signals:
            #for signal in frame.signals:
            #    if signal.multiplex is None:
            #        sig = createSignal(signal, nodeList, signalTypeEnums)
            #        message.append(sig)

ebroecker, 2018-07-26:
is this fixed with #174 ?!

ebroecker, 2018-08-16:
I think this is done already with the PR. Reopen if still fails

closed 2018-08-16 by ebroecker