ebroecker/canmatrix #177: dbc load function throws exception with version 0.6 but doesn't with version 0.5

miichiii, 2018-08-09:
Short issue-description:
----------------
We recently tried to use the latest version (0.6) of canmatrix and 
stumbled upon an issue inside the dbc.py load function. The issue 
revolves around the new segments that were added to that function. The 
regular expression which should match SIG_VALTYPE_ does not match 
anything within the dbc file which we're using (see attached files). By 
calling the groups function on that NoneType (no match) variable an 
exception gets thrown.

**However using version 0.5 which does not contain those newly inserted 
segments no exceptions are thrown.**

Detailed
----------

The problem might be within the regular expression:
```python
regexp = re.compile("^SIG\_VALTYPE\_ +(\w+) +(\w+) +\:(.*);")
temp = regexp.match(decoded)
frame = db.frameById(temp.group(1))
```
The regex expects a whitespace right in front of the colon.
In the case of our dbc file there is no whitespace inbetween the last 
word (``(\w+)``) and the colon (see below) which results in no match and in the ``temp``-variable 
being None:

``SIG_VALTYPE_ 103 demcnDuwOdvbNIyt: 1;``
``SIG_VALTYPE_ 103 RkriJGFLzdjmXJlt: 1;``
``SIG_VALTYPE_ 103 fpnbSRRONiJTjQST: 1;``

Upon calling the groups function of the ``temp``-variable the exception gets
thrown.

Potential workaround
----------------------

To make the function work I changed the regular expression to:

```python
regexp = re.compile("^SIG\_VALTYPE\_ +(\w+) +(\w+)\s*\:(.*);")
```

After switching out  **``(\w+) +\:``** with **``(\w+)\s*\:``**  no 
exception was thrown and the created object matched the object which was
created upon calling the load function in version 0.5.

To try out and analyse the problem yourself I provided the dbc file in 
the attachments.  

Attachments:
-------------
[bug_dbc.zip](https://github.com/ebroecker/canmatrix/files/2273307/bug_dbc.zip)

ebroecker, 2018-08-09:
Hi miichiii,

thank you very much for your really comprehensive bug report. Maybe one of the bug reports with the most relevant information in it.
At first view, I think I'll just add your suggestion to the development-branch. 
Tanks again

ebroecker, 2018-08-10:
should be fixed in development branch

closed 2018-08-10 by ebroecker