ebroecker/canmatrix #178: TypeError: 'NoneType' object is not iterable while converting xlsx -> dbc

shaolinkhoa, 2018-08-14:
Hi,

I just replaced a new computer which use win10. I installed python 3.6.4
I downloaded the canmatrix and installed it
I tested with my old excel input.xlsx then it return

```
D:\test>python convert.py input.xlsx aa.dbc
ERROR - formats - This file format is not supported for reading
Traceback (most recent call last):
  File "convert.py", line 6, in <module>
    canmatrix.convert.main()
  File "C:\Users\khoa\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\convert.py", line 393, in main
  File "C:\Users\khoa\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\convert.py", line 47, in convert
TypeError: 'NoneType' object is not iterable
```

I don't understand. I still can convert the input.xlsx in my old computer.
What did I miss ?

ebroecker, 2018-08-14:
Did you install xlrd?
`pip install git+https://github.com/ebroecker/canmatrix#egg=canmatrix[xlsx]`

shaolinkhoa, 2018-08-14:
Another error @@

```
D:\test>python convert.py template.xlsx test.dbc
ERROR - xlsx - xlsx: using legacy xlsx-reader - please get xlrd working for better results!
Traceback (most recent call last):
  File "convert.py", line 6, in <module>
    canmatrix.convert.main()
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\convert.py", line 393, in main
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\convert.py", line 41, in convert
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\formats.py", line 64, in loadp
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\formats.py", line 76, in load
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\xlsx.py", line 515, in load
TypeError: __init__() got an unexpected keyword argument 'dlc'
```

ebroecker, 2018-08-14:
I just saw now .

seams xls-reader makes errors and thus legacy-xlsx-reader is used. 
This is not nice because the legacy-xlsx-reader is very poor.

Ususally the xls-reader also reads xlsx-files. 
Could you try to rename your .xlsx in .xls (just for test). 
Than we will see why your xls-reader is not working correctly.

Thanx

ebroecker, 2018-08-14:
Did a further update for the legacy-reader. But this will probabpy not be the last update...
https://github.com/ebroecker/canmatrix/commit/2880c723221350c980d8eb93a521f6d85b5713fb

shaolinkhoa, 2018-08-15:
Hi, 

It has another error
```
D:\test>python convert.py template.xlsx new.dbc
ERROR - xlsx - xlsx: using legacy xlsx-reader - please get xlrd working for better results!
Traceback (most recent call last):
  File "convert.py", line 6, in <module>
    canmatrix.convert.main()
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\convert.py", line 393, in main
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\convert.py", line 41, in convert
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\formats.py", line 64, in loadp
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\formats.py", line 76, in load
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\xlsx.py", line 517, in load
AttributeError: 'list' object has no attribute 'addFrame'
```

Did you change the way input variables for template xlsx?

My solution: I installed the old version Commits on Jul 28, 2018 - The version that we talked last time: https://github.com/ebroecker/canmatrix/issues/165
And it works.

ebroecker, 2018-08-15:
as I already mentioned - you are using the legacy-reader:
`ERROR - xlsx - xlsx: using legacy xlsx-reader - please get xlrd working for better results!`

 The legacy-reader does not support any new features. This is why I just disabled it now.

The xls-reader (which is also the xlsx-reader) seems to make errors at your side. 
Could you please try again?

shaolinkhoa, 2018-08-16:
Hi,

I just installed the new update  **disable xlsx-legacy support, because it is outdated (#178)** 
But I still got
```
D:\test\canmatrix-be4fba6c07bba6ca468a80cc864130e74b14afce\examples>python convert.py template.xlsx test.dbc
Traceback (most recent call last):
  File "convert.py", line 6, in <module>
    canmatrix.convert.main()
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\convert.py", line 393, in main
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\convert.py", line 41, in convert
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\formats.py", line 64, in loadp
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\formats.py", line 76, in load
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\xlsx.py", line 444, in load
  File "C:\Users\ukn1hc\AppData\Local\Programs\Python\Python36\lib\site-packages\canmatrix-0+unknown-py3.6.egg\canmatrix\xls.py", line 399, in load
AttributeError: 'list' object has no attribute 'add'
```

It also happens with file.xls.

ebroecker, 2018-08-16:
@shaolinkhoa 
Thanks, seems our very small tests don't test 'adding' ecus from xls-format.
should be fixed

shaolinkhoa, 2018-08-16:
Thank you.
I installed the new commit e35e277 and it works.
Just one more thing. In the cmTemplate.xlsx , Launch Type column, you should add string "Cycle". If not, after convert xlsx -> dbc , the dbc file will be error and can't open.

ebroecker, 2018-08-16:
thanx again

closed 2018-08-16 by ebroecker