ebroecker/canmatrix #179: Running directly from the repo

kfurge-egi, 2018-08-16:
Is there a way to run the script directly from the repo without having to go through an installation step first?  I'm trying to debug an arxml import issue and being able to run the scripts directly would make the process more efficient.

altendky, 2018-08-16:
You should install in a virtualenv or venv, but there are a couple things you may find useful.  If you are editing the code as part of your testing then -e as a pip install argument.  If not then maybe installing via the URL.

git+http://GitHub.com/ebroecker/canmatrix@development#egg=canmatrix[arxml]

If you can share more then we can give more specific help.

altendky, 2018-08-17:
Mostly it's best to not use the system pip, even for 'system wide' installs.  You lose the ability to isolate activities.  Is this Linux or Windows?  In Linux when you --user pip is not in control of all the dependencies which isn't a great thing in some cases.

Anyways, I'm glad the -e was helpful.

kfurge-egi, 2018-08-17:
Yes, you're right.  That's good advice for others who may read this thread.  In my particular case, I have minimal pollution of the python package space, so I didn't have to worry too much about the finer points.

altendky, 2018-08-17:
Shall we close this or is there more on this topic?  If you have a specific issue with the arxml feel free to open another issue for that.  Of course, a sample file can be very helpful in that case.

kfurge-egi, 2018-08-17:
Yes, we can close this thread.  I'll post the actual issue I'm looking into separately.  Thanks again!

altendky, 2018-08-17:
Unless it's really obvious we probably won't be quite as quick, but eduard is usually pretty quick regardless.

closed 2018-08-17 by altendky