ebroecker/canmatrix #181: Bot nothing when convert arxml to dbc/xlsx

shaolinkhoa, 2018-08-21:
Hi,

I tried convert file.arxml ( autosar 4.2.1) to dbc/xlsx. 
```
python convert.py input.arxml output.dbc
python convert.py input.arxml output.xlsx
```
There is no error but it didn't output anything.

Do you know why ?

ebroecker, 2018-08-21:
please use on errors `python convert.py -vv input.arxml output.dbc` to find more information .

shaolinkhoa, 2018-08-21:
```
D:\python convert.py -vv input.arxml output.xlsx
INFO - convert - Importing input.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 0 frames in arxml...
DEBUG - arxml - DEBUG 0 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 0 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting output.xlsx ...
INFO - convert - done

D:\python convert.py -vv input.arxml output.dbc
INFO - convert - Importing input.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 0 frames in arxml...
DEBUG - arxml - DEBUG 0 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 0 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting output.dbc ...
INFO - convert - done
```
It seems good but I can't see any files in folder

gwallace41, 2018-10-02:
This looks like a similar issue.  Can you help?

INFO - formats - xls is not supported
INFO - formats - xlsx is not supported
INFO - formats - yaml is not supported
INFO - convert - Importing GM.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 547 frames in arxml...
DEBUG - arxml - DEBUG 547 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 4276 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
DEBUG - arxml - Busname: CAN_5_Cluster
DEBUG - arxml -  Speed: 500
DEBUG - arxml -  Speed: 500
DEBUG - arxml - 547 frames found in arxml

DEBUG - arxml - processing Frame: Tp_IPC_LCMFileDat_Data_CAN5_MSG_On_CAN_5_Cluster
DEBUG - arxml - IPC_LCMFileDat_Data_PDU
DEBUG - arxml - No Compmethod found!! - try alternate scheme.
Traceback (most recent call last):
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\wallgor\AppData\Local\Programs\Python\Python37-32\Scripts\canconvert.exe\__main__.py", line 9, in <module>
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\convert.py", line 294, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\convert.py", line 41, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\formats.py", line 54, in loadp
    return load(fileObject, importType, **options)
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\formats.py", line 64, in load
    dbs = moduleInstance.load(fileObject, **options)
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\arxml.py", line 1553, in load
    ns))
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\arxml.py", line 1315, in getFrame
    getSignals(signaltopdumaps, newFrame, arDict, ns, None)
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\arxml.py", line 1112, in getSignals
    is_float=is_float)
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\canmatrix.py", line 240, in __init__
    self.setMax()
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\canmatrix.py", line 433, in setMax
    self._max = self.calcMax()
  File "c:\users\wallgor\appdata\local\programs\python\python37-32\lib\site-packages\canmatrix\canmatrix.py", line 440, in calcMax
    return self._offset + (rawMax * self._factor)
OverflowError: int too large to convert to float

ebroecker, 2018-10-04:
@gwallace41
this is another issue, and this is a bug.
Did you use currend dev-branch?

ebroecker, 2018-11-21:
no more feedback at all - please open new issue, if still having issues with arxml

closed 2018-11-21 by ebroecker