ebroecker/canmatrix #189: Bug in symbol file generation

gbmont, 2018-08-27:
Hi,

Thank you for providing the CANmatrix-project.

I am using the project for generation of dbc and symbol files. If I try to open a symbol file in  PCAN Symbor editor V 6.2.0.1923 I get an error. After a little bit testing I found a possible problem in the sym.py-file.

Problem: Standard and extended message ID
If the length of a message ID contains leading zeros for a standard message type, 
PCAN Symbol editor can not read the file.

FormatVersion=5.0 // Do not edit this line!
Title="canmatrix-Export"

{SENDRECEIVE}

[DE_IO1_29]
ID=00000470h <== generates an error
Type=Standard  
DLC=4
Var=din_01 unsigned 0,1	// state of digital input
Var=din_02 unsigned 1,1	// state of digital input

Please have a look at
https://github.com/gbmont/canmatrix/blob/bugfix_version06/canmatrix/sym.py#L168

Best regards
Geri

altendky, 2018-08-27:
It looks like this was addressed in https://github.com/ebroecker/canmatrix/commit/8ab2bc648ebeef5f9c6b7368d260cd9c1b38a7fc for #139.  Are you still having trouble with latest from the `development` branch?

gbmont, 2018-08-28:
Okay your are totally right. Sorry for that.

closed 2018-08-28 by gbmont