ebroecker/canmatrix #191: Latest 'development' branch very slow reading/converting large arxml files

kfurge-egi, 2018-09-03:
This recent commit in 'development' has caused at least a 100x slowdown when converting a large (50 MiB) arxml file to dbc format:

```
commit 693df7d0eaa4c0667dc162f64f4807e63d146f07
Author: ebroecker <eduard@gmx.de>
Date:   Fri Aug 17 21:51:39 2018 +0200

    arxml update, use a more xpath like solution

```
On a local branch derived from the current 'development' branch, I reverted back to using the old arTree class and autosarhelpers.py and the performance returned.

kfurge-egi, 2018-09-04:
Here's some actual measurements on the same 50 MiB file:
```
real	15m30.566s  (930 seconds)
real	0m47.977s   (48 seconds)
```

ebroecker, 2018-09-06:
@kfurge-egi 
Thanks for your issue-report and your measurements!
I hoped to get better Performance with the "xpath"-Approach (especially because it's using some sort of cache) ...

I'll add an option for both arpath implementations.

kfurge-egi, 2018-09-11:
Works great, thanks.  As a side-note, the arxml file I have been working with has a bad path in it that is causing tree.xpath() in getArPath() to consume 100% cpu and never return.  Any thoughts on that?

ebroecker, 2018-09-11:
seems xpath-implementation is not the best in etree. 
I'll try to refactor the caching xpath approach to use etree.find. 
Maybe this makes it faster...

closed 2018-10-11 by ebroecker