ebroecker/canmatrix #193: NameError: name 'arDict' is not defined

msuleman67h, 2018-09-10:
```
Traceback (most recent call last):
  File "c:\program files\anaconda3\lib\runpy.py", line 193, in _run_module_as_ma
in
    "__main__", mod_spec)
  File "c:\program files\anaconda3\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Anaconda3\Scripts\canconvert.exe\__main__.py", line 9,
in <module>
  File "c:\program files\anaconda3\lib\site-packages\canmatrix\convert.py", line
 393, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "c:\program files\anaconda3\lib\site-packages\canmatrix\convert.py", line
 41, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "c:\program files\anaconda3\lib\site-packages\canmatrix\formats.py", line
 64, in loadp
    return load(fileObject, importType, key, flatImport, **options)
  File "c:\program files\anaconda3\lib\site-packages\canmatrix\formats.py", line
 74, in load
    dbs = moduleInstance.load(fileObject, **options)
  File "c:\program files\anaconda3\lib\site-packages\canmatrix\arxml.py", line 1
556, in load
    db.addFrame(getFrame(frameTrig,topLevelPackages,multiplexTranslation,ns, flo
at_factory))
  File "c:\program files\anaconda3\lib\site-packages\canmatrix\arxml.py", line 1
320, in getFrame
    getSignals(pdusigmapping, newFrame, xmlRoot, ns, None, float_factory)
  File "c:\program files\anaconda3\lib\site-packages\canmatrix\arxml.py", line 1
170, in getSignals
    temp = arGetChild(baseType, "SHORT-NAME", arDict, ns)
NameError: name 'arDict' is not defined
```

I get this error when converting arxml to dbc. Has anyone encountered this issue? 
Thank you

ebroecker, 2018-09-10:
Hi @msuleman67h 
seems to be fixes already in current development branch.
Could you try?

Thanks

msuleman67h, 2018-09-16:
> Hi @msuleman67h
> seems to be fixes already in current development branch.
> Could you try?
> 
> Thanks

Thank you. The issue is solved now.

closed 2018-09-16 by msuleman67h