ebroecker/canmatrix #200: Plan to discontinue py2 support someday?

altendky, 2018-10-03:
I am content to be helpful and deal with 2/3 compatibility to help people out...  but I can't say I'm excited about continuing to do so for a decade because others won't update.  Anyways, something to consider.  If discontinuing support obviously more notice to users is better.  Maybe we spin up a poll in here somehow?

ebroecker, 2018-10-04:
As usually you ask the right questions!
Meanwhile I can imagine to dismiss py2 compatibility.
My whish would be to get a more or less stable state of canmatrix with encoding and decoding working as expected. 
After releasing this as "1.0" I am very open for discarding any py2-support.
Than I would only fix heavy bugs in 1.0-py-branch but do implement new features only for py3...

what do you think?

A poll is also a good idea...

umlaeute, 2019-09-10:
i'm all for removing Py2-support  :-)

ebroecker, 2023-07-27:
from now on (version 1.0), no further support for py2...

closed 2023-07-27 by ebroecker