ebroecker/canmatrix #201: Missing some data during conversion of arxml to dbc

Jayanthdj, 2018-10-08:
When I tried to convert from arxml to dbc, some data are not captured as you can see below in the screen shot (viewed from arxml viewer)  and also I have listed it below which are all the attributes capturing those values.  Copying this manually is taking lots of time. Could you please help us to capture these data as well. Thank you so much in advance

[image]

[image]

[image]

<TIMEOUT> as Supervised Timeout
Timeperiod is periodicity
<MINIMUM-DELAY> is MinUpdate time
<CAN-ADDRESSING-MODE> is MsgIDType
</CAN-FD-FRAME-SUPPORT> is MsgType
</START-POSITION> IS PDU position
<LONG-NAME>
<L-4 L="EN">Engine Speed Protected : Engine Speed State Authenticated</L-4>
</LONG-NAME>     --> Long name (It is needed under an attribute called p_comment)

</TRANSFER-PROPERTY> as Transfered property --> value -->Triggered on change

<SHORT-NAME>StBltShfIntlkReqSts</SHORT-NAME>
              <PHYSICAL-PROPS>
                <SW-DATA-DEF-PROPS-VARIANTS>
                  <SW-DATA-DEF-PROPS-CONDITIONAL>
                    <COMPU-METHOD-REF DEST="COMPU-METHOD">/SDSIG_Hierarchy/SDSIG_PassiveSafety/StBltShfIntlkReqSts_CompuMethod</COMPU-METHOD-REF>  ---> CompuMethod

ebroecker, 2018-10-26:
sorry @Jayanthdj ,
currently I have not realy much time for digging deep into ARXML.
Also these attributes are not standard-dbc (or maybe FD-specific) attributes. Information about mapping/naming has to  be evaluated. Without example ARXML and dbc this is nearly not possible.
I guess this is a can-FD issue. I have no FD-ARXML around so no chance to help you now.

If possible, send me an example-ARXML and a fixed dbc (maybe as PN) than - maybe I could help.

Jayanthdj, 2018-10-26:
Hi, Edward. I have sent you example arxml file to your mail id eduard@gmx.de.

ebroecker, 2018-10-26:
hi @Jayanthdj , could you please resend, your mail was somehow deleted in spam and i cannot recover it...

Jayanthdj, 2018-10-26:
Hi Eduard,  I sent again.

ebroecker, 2018-10-29:
@Jayanthdj 
Seems the email still does not arrive at my side. I've no idea what's going wrong.

Jayanthdj, 2018-11-05:
I think the file size was more. I have compressed it and sent you now. Please check :)

ebroecker, 2018-11-05:
sorry - again no email here. 
I changed something at my side now. Could you please resent one more time?
Thanks

ebroecker, 2018-11-05:
Thanks,
this time it worked. (Seems I have some issue with my mail-server, have to check afterwards).

Jayanthdj, 2018-11-05:
Great! Please help me in capturing all other values which are mentioned and Thank you so much in advance :)

ebroecker, 2019-02-23:
should be fixed, otherwise please create a new ticket or reopen

closed 2019-02-23 by ebroecker