ebroecker/canmatrix #222: Fibex Import missing

ebroecker, 2018-10-26:
I have a problem with Converting a Fibex file into an Yaml format using Canmatrix open source code.

Traceback (most recent call last):
  File "convert.py", line 304, in <module>
    sys.exit(main())
  File "convert.py", line 301, in main
    convert(infile, outfileName, **cmdlineOption
  File "convert.py", line 51, in convert
    for name in dbs:
TypeError: 'NoneType' object is not iterable

I found that in Fibex,py file Load function is not defined is it not supported by Canmatrix ?

Please let me know thanks in advance.

Regards,
Ibrar

_Originally posted by @ibrarmohammed in https://github.com/ebroecker/canmatrix/issues/90#issuecomment-433376419_

ebroecker, 2018-10-26:
@ibrarmohammed 

fibex import not supported yed. 
I even did not hear anybody using the fibex export. 

Keep in mind - even export ONLY supports fibex-CAN export.

ibrarmohammed, 2018-10-26:
Thanks a lot lot , But I think we have requirement to check Fibex file and export in Yaml format.

ebroecker, 2018-11-21:
sorry, 
noone here who wants/is able to implement a fibex importer/loader

ebroecker, 2023-01-21:
@ibrarmohammed 

so now we have some (CAN-)fibex import implemented in branch: https://github.com/ebroecker/canmatrix/tree/fibex_read

I guess you don't need it anymore :-)

ibrarmohammed, 2023-01-21:
Thanks Eduard , Yes Right now  not As i am implementing my own to parse
fibex but I will have look into it .

Regards,
Ibrar

ebroecker, 2023-01-23:
would you need CAN only or also Flexray/LIN?

ibrarmohammed, 2023-01-23:
CAN only for now.

ebroecker, 2023-01-23:
Hi @ibrarmohammed 

CAN works (at least with the example fibex-file I received).
Would be interesting if it fulfills your needs also... (But remembre, for now you'd have to use the branch: https://github.com/ebroecker/canmatrix/tree/fibex_read)

ibrarmohammed, 2023-01-23:
Thanks for the tip!

closed 2023-06-22 by ebroecker