ebroecker/canmatrix #226: Issue in converting arxml to dbc

Jayanthdj, 2018-10-29:
When I tried to convert .arxml to dbc file, I got this below error
[image]

ebroecker, 2018-10-29:
Initvalue translation is buggy.

https://github.com/ebroecker/canmatrix/blob/master/canmatrix/arxml.py#L1140-L1145

hard coded lable to value translation fails if lable is something other than "true" or "false"

maybe this lookup is not bullet prove:
https://github.com/ebroecker/canmatrix/blob/master/canmatrix/arxml.py#L1051

Jayanthdj, 2018-10-30:
Hi,
After doing the modification you suggested, I am getting this below error
if initvalue is None or initvalue.__len__() == 0:
                initvalue = arGetXchildren(isignal, "INIT-VALUE/NUMERICAL-VALUE-SPECIFICATION/VALUE", arDict, ns) ##AR4.2
            elif initvalue is not None and initvalue.__len__() >= 1:
				initvalue = initvalue[0]

[image]

Jayanthdj, 2018-11-05:
I tried to convert arxml to dbc  with 'Canmatrix development' and got this below error
[image]

and with Canmatrix master I got this below error  :(
[image]

ebroecker, 2018-11-05:
could you - maybe - provide an example also? PN?

Jayanthdj, 2018-11-05:
Hii, I sent

ebroecker, 2018-11-05:
Thanks,

not sure, your ARXML is correct:
```xml
<INIT-VALUE>
                <NUMERICAL-VALUE-SPECIFICATION>
                  <VALUE>On</VALUE>
```

in my unserstanding "On" is not realy numerical. But maybe this would be translated by the compuMethod '/CompuMethods/OnOff1_1'. But as far as I can see, this compuMethod is not included in the ARXML.

I'll provide a patch for fixing On and Off - but not sure if its the right way to do it...

ebroecker, 2018-11-21:
merged

closed 2018-11-21 by ebroecker