ebroecker/canmatrix #229: DBC: Parsing of VAL_ when VAL_TABLE_ name is supplied

msalau, 2018-10-31:
Hello,

Please clarify usage of the VAL_TABLE_ feature.
I used description of VAL_ and VAL_TABLE_ from here: https://github.com/stefanhoelzl/CANpy/blob/master/docs/DBC_Specification.md#val_

According to it the following snippets should be the same:
```
VAL_ SG_ 1 Sig1 0 "VAL_0" 1 "VAL_1";
```
```
VAL_TABLE_ Sig1_Values 0 "VAL_0" 1 "VAL_1";
VAL_ SG_ 1 Sig1 Sig1_Values;
```

The first approach is supported by `canmatrix`, but the second is not.
VAL_TABLE_ is parsed and imported, but the signal is not associated with it.
Here is the snippet that reads VAL_
https://github.com/ebroecker/canmatrix/blob/74ba95dfa664fd0a7722f97de9ad9944de6825fb/src/canmatrix/dbc.py#L691
It only expects inline values, not the value table name.

Please clarify if I understand this correctly.

Thanks an regards,
Maksim.

ebroecker, 2018-10-31:
First of all, you are right, only the first approach is supported.

by the way: the first approach seems to have a typo:
```
VAL_ SG_ 1 Sig1 0 "VAL_0" 1 "VAL_1";
```
imho it should look like this (without 'SG_'):
```
VAL_ 1 Sig1 0 "VAL_0" 1 "VAL_1";
```
---

It wouldn't be a big deal to support, but I can't get the second approach work with candb++:
```
VAL_TABLE_ Sig1_Values 0 "VAL_0" 1 "VAL_1";
...
VAL_ SG_ 1 Sig1 Sig1_Values;
```
I also find no reference for this syntax [link](https://edoc.site/queue/dbc-format-2007-pdf-free.html) so maybe [CANpy](https://github.com/stefanhoelzl/CANpy/blob/master/docs/DBC_Specification.md#val) has special syntax extension? I don't know...

msalau, 2018-11-01:
> by the way: the first approach seems to have a typo:

I wrote this by hand as an example and didn't verified in CANdb++ editor :(

> It wouldn't be a big deal to support, but I can't get the second approach work with candb++:

Indeed. I've tried it in CANdb++ and it didn't work as I hoped for.
I created a table, associated it with a signal and saved the resulting file. It appeared that the editor inlined values into VAL_. On the subsequent load the table was not associated with the table any more thus leaving the table as a reference for future edits, but actually unused in the current version.

Thanks!

closed 2018-11-01 by msalau