ebroecker/canmatrix #230: Convert DBC to json - only 32 characters of Signal Name possible?

jehoff, 2018-10-31:
When I convert one of our .dbc files into json format, the signal names are cut. 
They are too long. I know Vector Informatik wants the signal names not longer than 32 characters. But .dbc files work with signal names longer than 32 characters.

Would it be possible to change this limitation to 64 characters in canmatrix?
Or maybe somebody can tell me where to change this limitation of 32 characters in the code?

I like canmatrix, but i cant separate/find signals with this limitation.

When i convert with canmatrix i got this in example:
"name": "FAILURE_ZELL_UNTERTEMPERATU_0042"

in .dbc file it could be:
"FAILURE_ZELL_UNTERTEMPERATUR_ENTLADEN_ALARM_IDX_01"
or
"FAILURE_ZELL_UNTERTEMPERATUR_ENTLADEN_WARNUNG_IDX_01"
or
"FAILURE_ZELL_UNTERTEMPERATUR_LADEN_ALARM_IDX_01"
or
"FAILURE_ZELL_UNTERTEMPERATUR_LADEN_WARNUNG_IDX_01"

Is it possible to change the limitation of 32 characters per signal name?

altendky, 2018-10-31:
Looks ok on the dbc import side.  I'll add a long signal name export test for json.

altendky, 2018-10-31:
Alrighty, output alone tests ok as well.  @jehoff what version of canmatrix are you using?  How did you install it?  It's generally best to install from git at this point as opposed to pypi.  Also, what exactly are you doing to do the conversion?  If you can provide the actual input and output files that would be great, though I understand that's often proprietary data.

https://github.com/ebroecker/canmatrix#install-instructions

But, I'll check out the convert cli next.

altendky, 2018-10-31:
well, on pypy they worked.  fixing to work on all interpreters (or find an issue).

ebroecker, 2018-10-31:
My stupid tests with "real" dbc also worked - no 32 character limitation.
I even cannot remember for a version of canmatrix, with an 32-character limitation.

@jehoff 
could you look into the dbc with a text editor?
Maybe there are the "shortend" signals in the dbc, but some other attribute which holds the complete name?
search for  "FAILURE_ZELL_UNTERTEMPERATU_0042" in the dbc and you may find
SG_ FAILURE_ZELL_UNTERTEMPERATU_0042.

search for "FAILURE_ZELL_UNTERTEMPERATUR_ENTLADEN_ALARM_IDX_01" and you will find some "attribute", which holds the long name.
I have no candb++ for double checking this issue.

jehoff, 2018-11-02:
@altendky @ebroecker  Thank you for your helfp. I checked the .dbc in an editor. Your are right, the signal names are cut in editor (notepad++)! When i open the .dbc in CANdb Editor the signal names are not cut. 
The problem is the .dbc file, ... too long signal names.
But there is an attribute, or an explanation in the .dbc file when you open it with an editor:
> BA_ "SystemSignalLongSymbol" SG_ 193 FAILURE_ZELL_UNTERTEMPERATU_0043 "FAILURE_ZELL_UNTERTEMPERATUR_LADEN_WARNUNG_IDX_01";

This is the Signal name in .dbc
> SG_ FAILURE_ZELL_UNTERTEMPERATU_0043 : 13|1@1+ (1,0) [0|1] "Warnung"  BMS_MASTER

I think CANdb Editor is working with this, to show the long signal name.

I use canmatrix 0.6, installed with pip on Windows 7
Thanks for your fast reply!

Notepad (Screenshot)
[image]

CANdb Editor (Screenshot)
[image]

Notepad (Attribute SystemSignalLongSymbol)
[image]

ebroecker, 2018-11-02:
@jehoff

Thanks for your Analysis.
I'll create a fix for this issue!

ebroecker, 2018-11-05:
simmilar issue exists for:
* frame names
* enviroment variable names
* ECU names

PR #293 (2019-02-22): [WIP] Iss230 support for long names 
should fix #230
commit 04b9e2ad (2019-02-22): [WIP] Iss230 support for long names  (#293)

fixes #230 
parts from #236
* add support for long ECU names
* fix long enviroment names
refactor: remove import *
refactor: create small sub-functions for dbc creation
closed 2019-02-22 by ebroecker