ebroecker/canmatrix #237: Use loggers based on __name__

Funth0mas, 2018-11-02:
The canmatrix project tends to be more than a standalone converter only. I'd try to refactor loggers to use getLogger(__name__) instead of the root logger everywhere. The `setup_logger` would configure only top-level `canmatrix` logger. 
https://docs.python.org/2/howto/logging.html#configuring-logging-for-a-library.

altendky, 2018-11-03:
At least part of this got stalled in #164.  If we still like that we can revive it and get it merged then follow up with wherever else it needs tidied?

Funth0mas, 2018-11-03:
Yes I forgot to state that this is inspired by the mentionned PR, but it tried to fix single file only and the diff currently includes about three lines of code (among not being mergable currently).
I would do it globally (I don't know in how many files the logger is actually used). That is why I'm "asking" before working on it.

altendky, 2018-11-03:
I agree that `__name__` should generally be used. Perhaps in entry point functions it is proper to use `None` (explicitly or implicitly, no matter) and do the basic config or whatever?

ebroecker, 2018-11-21:
done?!

closed 2018-11-21 by ebroecker