ebroecker/canmatrix #238: j1939 Frame setters might raise TypeError

Funth0mas, 2018-11-02:
The j1939 setters (pgn, priority, source) raise TypeError exception, originating in `recalcJ1939Id()`, if all properties weren't filled already. I don't know the use case but maybe add a guard to recalcJ1939Id like `if None in (self.j1939_pgn, self.j1939_prio, self.j1939_source): return` ?

ebroecker, 2018-11-02:
@Funth0mas ,
yes - a guard is a good idea.

altendky, 2018-11-05:
Each of the three (source/pgn/priority) correspond to independent sections of the 29-bit identifier, yes?  So setting any one of them individually shouldn't depend on the others being specified.  The only uncertainty would be if the identifier itself were, say, `None` then you set the PGN, should that raise an exception?  Fill in 0 for the non-PGN bits and set the PGN bits properly?  I think that source/pgn/priority might not need to have their own storage at all but rather just encoding into the identifier and decoding it out.  This would also eliminate the concept of recalculating either direction.

I might tend to basically let the natural exceptions drive our behavior decision, even if we decide to raise more descriptive errors.  For example, if the identifier is `None` then setting any of source/pgn/priority would fail via exception.  If you were to try to read any of them with a `None` identifier it would raise as well.  I would probably add exceptions in case they are read or written with a standard 11-bit identifier selected.

commit d5c5f530 (2019-08-19): fix for #238 (j1939 Frame setters might raise TypeError) (#389)

move j1939 (pgn, prio, source) handling direct to arbitration-id class
(fixes #238)
commit 889c2952 (2019-11-08): merge to master for release 0.9 (#429)

* include LICENSE in source distribution (#365)
* [arxml]: rx/tx typo fix 
* annotate and fix codestyle dbc (#369)
* Always use Signal.is_little_endian as bool #326 (#371)
* Add coding, remove #!python where inappropriate (#370)
* annotation cleanup ticket #323 (#372)
* Update cli doc (inspired by #361) (#373)
* fix: xls is using the wrong number for arbitration id (#377)
* add outputformat for Scapy (#378)
* fix special char handling in scapy exporter
* Dump factor, offset, min, max as float. Load as numeric (#380)
* added --jsonNativeTypes as cli optio [... cut]
commit 63c2f35d (2020-09-29): prepare release (#512)

* include LICENSE in source distribution (#365)
* arxml: rx/tx typo fix 
* arxml ipdu-groups fix
* annotate and fix codestyle dbc (#369)
* Always use Signal.is_little_endian as bool #326 (#371)
* Add coding, remove #!python where inappropriate (#370)
* annotation cleanup ticket #323 (#372)
* Update cli doc (inspired by #361) (#373)
* fix: xls is using the wrong number for arbitration id (#377)
* add outputformat for Scapy (#378)
* simplify get_fmt
* Dump factor, offset, min, max as float. Load as numeric (#380)
* added --jsonNativeTypes as cli option
* ARXML: merge cont [... cut]
commit eb68ec11 (2022-01-11): merge for release 0.9.4 (#621)

* include LICENSE in source distribution (#365)

* arxml: rx/tx typo fix 

fix for #366

* annotate and fix codestyle dbc (#369)

* Always use Signal.is_little_endian as bool #326 (#371)

* Add coding, remove #!python where inappropriate (#370)

* annotation cleanup ticket #323 (#372)

* Update cli doc (inspired by #361) (#373)

* fix: xls is using the wrong number for arbitration id (#377)

* add outputformat for Scapy (#378)

* add output format for scapy
* fix special char handling in scapy exporter
* add unit test
simplify get_fmt

* Dump factor, offset, min [... cut]
closed 2019-08-19 by ebroecker