ebroecker/canmatrix #242: VFrameFormat

rpkvt, 2018-11-07:
Hello!
I'm writing software to create dbc files using canmatrix.  I'm using version f9b84b834b1b93d849823a01d4bcea33b01f8866

When I run my software I generated a dbc that looks something like this:

VERSION "created by canmatrix"

NS_ :

BS_:

BU_: 

BO_ 2566849152 Node415_0xff02: 14 Vector__XXX
 SG_ ch10 : 7|32@0- (1,0) [-2147483648|2147483647] "" Vector__XXX
 SG_ ch11 : 39|32@0- (1,0) [-2147483648|2147483647] "" Vector__XXX
 SG_ ch12 : 71|32@0- (1,0) [-2147483648|2147483647] "" Vector__XXX
 SG_ Padding_0 : 103|8@0+ (1,0) [0|255] "" Vector__XXX
 SG_ Padding_1 : 111|8@0+ (1,0) [0|255] "" Vector__XXX

BA_DEF_ BO_ "VFrameFormat" ENUM "StandardCAN","ExtendedCAN","StandardCAN_FD","ExtendedCAN_FD","J1939PG";
BA_DEF_ "ProtocolType" STRING;

BA_ "ProtocolType" "J1939";

BA_ "VFrameFormat" BO_ 2566849152 4;

Which gets interpreted like this:
[image]

Which doesn't seem to tell the receiving system how to parse the incoming datastream:
[image]

I have an example dbc file which looks like this (and seems to be parsed correctly):
[image]

If I change a line from the code above to this (changed the 4 to a 3):
BA_ "VFrameFormat" BO_ 2566849152 3;

Then the software seems to interpret it correctly:
[image]

The software shows the values of individual variables: 
[image]

My question is how do I get Canmatrix to output a 3 instead of a 4?

Here are the code blocks I wrote for creating frames and making the DBC:

```python
    def CreateFrame(self,msgName,pgn,priority=0x06,message=""):
        if len(self.signals) == 0:
            return
        frame = canmatrix.Frame(    msgName,
                                    j1939_pgn=pgn,
                                    j1939_prio=priority,
                                    j1939_source=self.src_addr,
                                    comment=message
                                )
        for sig in self.signals:
            frame.addSignal(sig)

        self.cm.frames.addFrame(frame)
        #self.cm.addFrame(frame)
        self.startBit = 0
        self.signals.clear()

    def CommitDBC(self,filename):
        self.cm.recalcDLC("force")

        print(canmatrix.formats.supportedFormats)

        # save dbc
        canmatrix.formats.dumpp({"": self.cm}, filename)
```

Thanks,
Rich

ebroecker, 2018-11-08:
Hi @rpkvt ,
Thanks for your Report.

I cannot reproduce the views in candb. 
You are creating a J1939-dbc. 
My knowledge about them is very limited. Issues witch canmatrix regarding J1939 are very likely.

In my eyes a VFrameFormat 'J1939PG' (which is '4') looks right. A '3' represents 'ExtendedCAN_FD'. 
But if I open the dbc, and do some not relevant changes, and save it again, a '3' is set. 
Seems you are right.

Another idea:
candb++ expects the enumeration to be like this:
```
"VFrameFormat" ENUM  "StandardCAN","ExtendedCAN","reserved","J1939PG"
```
and does not honor our self made enumeration. 
A quick check seems to verifiy my idea.

the code responsible for the frame-format (dbc attribute "VFrameFormat") is here: 
https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/dbc.py#L81-L101

I think, it is not possible to create a dbc with FD and J1939 mixed, because candb++ does not honor it. I'll have to investigate a bit further.

a quick fix for you could be to exchange the line https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/dbc.py#L88:
```
"VFrameFormat" ENUM  "StandardCAN","ExtendedCAN","StandardCAN_FD","J1939PG","ExtendedCAN_FD";
```
Positions of "J1939PG" and "ExtendedCAN_FD" are switched.

rpkvt, 2018-11-09:
@ebroecker 
Thanks for your response.  Yeah, I was wondering if that was the problem. Switching the two seemed to do the trick.

commit f1af9f08 (2019-08-19): DBC: fix #242. J1939 and FD Frames in dbc

candb seems to be queasy about the order of the VFrameFormat attribute
definition. This Patch seems to generate a dbc, which can contain can_fd
AND J1939-frames.
commit 889c2952 (2019-11-08): merge to master for release 0.9 (#429)

* include LICENSE in source distribution (#365)
* [arxml]: rx/tx typo fix 
* annotate and fix codestyle dbc (#369)
* Always use Signal.is_little_endian as bool #326 (#371)
* Add coding, remove #!python where inappropriate (#370)
* annotation cleanup ticket #323 (#372)
* Update cli doc (inspired by #361) (#373)
* fix: xls is using the wrong number for arbitration id (#377)
* add outputformat for Scapy (#378)
* fix special char handling in scapy exporter
* Dump factor, offset, min, max as float. Load as numeric (#380)
* added --jsonNativeTypes as cli optio [... cut]
commit 63c2f35d (2020-09-29): prepare release (#512)

* include LICENSE in source distribution (#365)
* arxml: rx/tx typo fix 
* arxml ipdu-groups fix
* annotate and fix codestyle dbc (#369)
* Always use Signal.is_little_endian as bool #326 (#371)
* Add coding, remove #!python where inappropriate (#370)
* annotation cleanup ticket #323 (#372)
* Update cli doc (inspired by #361) (#373)
* fix: xls is using the wrong number for arbitration id (#377)
* add outputformat for Scapy (#378)
* simplify get_fmt
* Dump factor, offset, min, max as float. Load as numeric (#380)
* added --jsonNativeTypes as cli option
* ARXML: merge cont [... cut]
closed 2019-08-19 by ebroecker