ebroecker/canmatrix #246: Missing function in ARXML.py

PierQuerio, 2018-11-19:
Hi,

While checking for the script (I wrote something similar for my personal usage and I am interested on your converter), I encountered the following issue:
 - At line 1654, the ARXML.PY file calls the function "computeSignalValueInFrame(...)". This function is NOT FOUND, it doesn't exist in the delivered package at all.

Can you have a look on it?

Regards,
Pier Querio

ebroecker, 2018-11-19:
Hi @PierQuerio ,
yes you are right, in the development branch this is still missing.

you could habe a Look on branch iss226_arxml_valueGuessing (https://github.com/ebroecker/canmatrix/commits/iss226_arxml_valueGuessing)

It's orinally another issue but this bug should be fixed there already.

PierQuerio, 2018-11-19:
Hi @ebroecker ,

Using the mentioned branch, the issue is gone and the DBC file(s) generated.
Thanks

Will you import the fix in the development branch, too?
Regards

ebroecker, 2018-11-21:
merged with 9dfd08efe39980be459a7377e4f28da9b07efc7d

closed 2018-11-21 by ebroecker