ebroecker/canmatrix #250: Create dbc file

gbmont, 2018-11-21:
Hello

I am working with Motorola data format and would like to add following signals to a frame.

Signal name	length
din01	 1 (motorola)
din02	 3 (motorola)
din03	 1 (motorola)
din04	 1 (motorola)
din05	 1 (motorola)
din06	 1 (motorola)
din07	 1 (motorola)
din08	 1 (motorola)
din09	 1 (motorola)
din10	 1 (motorola)
din11	 1 (motorola)
din12	 1 (motorola)
din13	 1 (motorola)

When I create a frame object and add the signals, 

My expected dbc-output should be:

``` file
 
VERSION "created by canmatrix"

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_: DE_29

BO_ 1136 DE_IO1_29: 2 DE_29
 SG_ d_in01 : 0|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in02 : 3|3@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in03 : 4|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in04 : 5|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in05 : 6|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in06 : 7|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in07 : 8|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in08 : 9|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in09 : 10|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in10 : 11|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in11 : 12|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in12 : 13|1@0+ (1,0) [0|1] "" Vector__XXX
 SG_ d_in13 : 15|2@0+ (1,0) [0|1] "" Vector__XXX

CM_ BO_ 1136 "Input-object 1, 32 digital inputs 01 to 32";
CM_ SG_ 1136 d_in01 "State of digital input 01";
CM_ SG_ 1136 d_in02 "State of digital input 02";
CM_ SG_ 1136 d_in03 "State of digital input 03";
CM_ SG_ 1136 d_in04 "State of digital input 04";
CM_ SG_ 1136 d_in05 "State of digital input 05";
CM_ SG_ 1136 d_in06 "State of digital input 06";
CM_ SG_ 1136 d_in07 "State of digital input 07";
CM_ SG_ 1136 d_in08 "State of digital input 08";
CM_ SG_ 1136 d_in09 "State of digital input 09";
CM_ SG_ 1136 d_in10 "State of digital input 10";
CM_ SG_ 1136 d_in11 "State of digital input 11";
CM_ SG_ 1136 d_in12 "State of digital input 12";
CM_ SG_ 1136 d_in13 "State of digital input 13";

```

..e.g. with d_in06 at position byte 0, bit 0 (from left to right).

Please let me know how to create this output.

Thank you

Geri

ebroecker, 2018-11-21:
```python
import canmatrix.formats
import canmatrix

cm = canmatrix.CanMatrix()

frame = canmatrix.Frame("DE_IO1_29", id = 0x470, comment = "Input-object 1, 32 digital inputs 01 to 32", transmitters = ["DE_29"], size = 2)

for i in range(1,14):
	signal = canmatrix.Signal("d_in%02d" % (i), is_little_endian = False, is_signed = False, size = 1, comment = "State of digital input %02d" % (i))
	signal.setStartbit(i+1, bitNumbering = True)
	frame.addSignal(signal)

frame.signalByName("d_in02").size = 3
frame.signalByName("d_in01").setStartbit(0, bitNumbering = True)
frame.signalByName("d_in13").size  = 2
frame.signalByName("d_in13").setStartbit(15, bitNumbering = True)
cm.addFrame(frame)

cm.updateEcuList()

canmatrix.formats.dumpp({"":cm}, "test.dbc")
```

gbmont, 2018-11-22:
Thank you very much Eduard for providing the example code! 

Now it works!

I had to add two lines of code (see hint in code). Maybe that helps some othters.

```python
import canmatrix.formats
import canmatrix

cm = canmatrix.CanMatrix()

frame = canmatrix.Frame("DE_IO1_29", id = 0x470, comment = "Input-object 1, 32 digital inputs 01 to 32", transmitters = ["DE_29"], size = 2)

start_bit = 0
for v in l.variables:  # read a list of objects containing sorted signal definitions
	extras = {}
	temp_signal = Signal(v.name,  startBit=start_bit, size=v.length, is_little_endian=(l.byte_order == 1),
		is_signed=(v.data_type == "Signed"), factor=v.factor, offset=v.offset, min=v.min,	max=v.max,
		unit=v.unit, receiver=[],	 **extras)

        # following two lines of code have been added to get the right bit numbering
	if not temp_signal.is_little_endian:  # is big endian, Motorola format
		temp_signal.setStartbit(start_bit + v.length - 1, bitNumbering=True)

	start_bit = start_bit + v.length
	frame.addSignal(temp_signal)

cm.addFrame(frame)
cm.updateEcuList()
canmatrix.formats.dumpp({"":cm}, "test.dbc")
```

closed 2018-11-22 by gbmont