ebroecker/canmatrix #251: Error when value or length not int

timschneider, 2018-11-21:
https://github.com/ebroecker/canmatrix/blob/9dfd08efe39980be459a7377e4f28da9b07efc7d/src/canmatrix/canmatrix.py#L531

This will actual fail if value or length isn't of type int. For me it was the case that length was Decimal, i fixed it by parsing the values as int() on that line.

Dunno if this is correct in the whole project, but for my case, it is.

Bye

ebroecker, 2018-11-21:
@timschneider 
Thanks for your report!

not sure how to handle this, but gonna think about.
@altendky should we cast to int in the `pack_bitstring` or force the user to use int?

timschneider, 2018-11-21:
@ebroecker you're welcome, as it sounds like from your suggestion, this is not a 'user' fault. this happend to me while parsing an dbc file and reassemble can messages from it.

ebroecker, 2018-11-21:
@timschneider 
I can't find an issue where dbc creates 'Decimal' for the.

Actual we do something similar in our test.

https://github.com/ebroecker/canmatrix/blob/b21a9ed3c17ab90dff53375a9c42ac5f75d321bf/src/canmatrix/tests/test_frame_decoding.py#L9-L26

could you please give some hint where canmatrix or dbc readers goes wrong...
Thanks

timschneider, 2018-11-22:
@ebroecker @altendky 
i could finally narrow it down to 

https://github.com/ebroecker/canmatrix/blob/b21a9ed3c17ab90dff53375a9c42ac5f75d321bf/src/canmatrix/canmatrix.py#L298-L306

But first what i'm doing is the following:

- Open DBC File
- Search for a frameByName
- Create a dict of Signals with setMin() as there values
- change the values I need to change
- encode the Frame by passing the dict to the encode function
- Send the Message via CAN

And there is the Problem, because setMin() will always(?) create a Decimal representation of the values, even for signals which are not float.

So maybe i'm using the interface wrong?

Tim

ebroecker, 2018-11-22:
@timschneider 
Thanks for thedetails.
In deed, this is an inconsistent behaviour of canmatrix. 

Not sure yet what's the right way to fix.
Gonna discuss this here...

altendky, 2018-11-23:
There's something...  more of the mess with partially initialized Signal instances and attribute order dependence.  I think at some point we'll see the light on how to handle this 'float factory' stuff.  I don't really like this inconsistent type situation so, I dunno.

ebroecker, 2019-08-14:
@altendky 
does #258 fix this issue, or should this issue kept open?

altendky, 2019-08-14:
A quick look over it seems to suggest it would fix the issue.

closed 2019-08-14 by altendky