ebroecker/canmatrix #257: Unable to disable report_error in decode.

timschneider, 2018-11-22:
https://github.com/ebroecker/canmatrix/blob/b21a9ed3c17ab90dff53375a9c42ac5f75d321bf/src/canmatrix/canmatrix.py#L1016

Without an additional parameter we should disable report_error by default.

altendky, 2018-11-23:
I agree with an exception over a print, but why should errors pass silently by default?

timschneider, 2018-12-03:
no sorry, this was just an intuitive way of reporting this "situation" like, oh, wait, no I see an output on the screen, what the heck? oh, ok it’s a debug output, but wait, how can I handle it? ummm, no way, cause it's a print - ok then turn it off and check it upfront. I don’t want to pass it silently. 

So I think an exception is the way we should go.

ebroecker, 2019-02-20:
fixed

closed 2019-02-20 by ebroecker