ebroecker/canmatrix #263: Errors while parsing DBC with comma in enum attribute values

blem14, 2018-11-29:
Errors occur while parsing DBC with `,` as one of enum attribute values.
It is recognized as valid option by Vector CANdb++.

Occurs on develop (ed4a1d5c80d0a1a4e48fa8a8492a3a913f95c365), attributes are not collected.
On master (f9b84b834b1b93d849823a01d4bcea33b01f8866) errors does not occur, attributes are collected, but with invalid values - `['"Val1"', '"', '"']`.

Expected to get `['Val1', ',',]`

Errors:
> error with line no: 37
> b'BA_DEF_ "example0" ENUM "Val1",",";\r\n'
> error with line no: 38
> b'BA_DEF_ BO_ "example1" ENUM "Val1",",";\r\n'
> error with line no: 39
> b'BA_DEF_ SG_ "example2" ENUM "Val1",",";\r\n'
> error with line no: 41
> b'BA_DEF_ BU_ "example4" ENUM "Val1",",";\r\n'

Can be reproduced with DBC containig these lines:
> VERSION ""
> 
> 
> NS_ : 
> 	NS_DESC_
> 	CM_
> 	BA_DEF_
> 	BA_
> 	VAL_
> 	CAT_DEF_
> 	CAT_
> 	FILTER
> 	BA_DEF_DEF_
> 	EV_DATA_
> 	ENVVAR_DATA_
> 	SGTYPE_
> 	SGTYPE_VAL_
> 	BA_DEF_SGTYPE_
> 	BA_SGTYPE_
> 	SIG_TYPE_REF_
> 	VAL_TABLE_
> 	SIG_GROUP_
> 	SIG_VALTYPE_
> 	SIGTYPE_VALTYPE_
> 	BO_TX_BU_
> 	BA_DEF_REL_
> 	BA_REL_
> 	BA_DEF_DEF_REL_
> 	BU_SG_REL_
> 	BU_EV_REL_
> 	BU_BO_REL_
> 	SG_MUL_VAL_
> 
> BS_:
> 
> BU_:
> 
> 
> 
> 
> BA_DEF_  "example0" ENUM  "Val1",",";
> BA_DEF_ BO_  "example1" ENUM  "Val1",",";
> BA_DEF_ SG_  "example2" ENUM  "Val1",",";
> BA_DEF_ EV_  "example3" ENUM  "Val1",",";
> BA_DEF_ BU_  "example4" ENUM  "Val1",",";
> BA_DEF_DEF_  "example0" ",";
> BA_DEF_DEF_  "example1" ",";
> BA_DEF_DEF_  "example2" ",";
> BA_DEF_DEF_  "example3" ",";
> BA_DEF_DEF_  "example4" ",";
> 
>

ebroecker, 2018-11-29:
@blem14 
Thank you very much for your bug report.

Reports of this quality are easy for us to work on, because you provided everything we need to reproduce.

I'll just no start to fix this issue.

ebroecker, 2018-12-04:
@blem14 
could you try branch `iss263`. 
should fix this issue.

`pip install git+https://github.com/ebroecker/canmatrix@iss263` should do it...

blem14, 2018-12-05:
This fixes the issue for comma, however further tests with other special characters shows that space is causing error too, not sure if it was like that earlier.

I used following line to test special character compatibility within CANdb++:

> BA_DEF_ BO_  "example" ENUM  "Val 1","vector_leerstring",""," ","'","(",")","[","]","/","-","|","{","}",";",":","<",">",".","?","!","@","#","$","%","^","&","=","`","~";

Here is screenshot of the given line's option list in the CANdb++.
As you can see, "vector_leerstring" is used by the CANdb++ to store information about empty option. Upon opening file "" is also recognized as valid option, only to be replaced with "vector_leerstring" if DBC is saved.

Fun fact - CANdb++ allows multiple options of the same string, but I haven't tested it yet.

[image]

ebroecker, 2018-12-05:
@blem14 
Thanks for providing even more test data.
I'll work on it.

ebroecker, 2018-12-05:
@blem14 
should be fixed now also

ebroecker, 2018-12-10:
@blem14 
No problem, ..., gonna fix this soon

ebroecker, 2018-12-10:
fixed

ebroecker, 2018-12-10:
fixed

closed 2018-12-10 by ebroecker