ebroecker/canmatrix #267: Cannot properly convert .sym to .xlsx or others with double inside .sym

DanielTongAwesome, 2018-12-03:
**I have a .sym file, and when I run the command:**   

**#canconvert   -v x.sym x.xlsx**

**#canconvert   -v x.sym x.dbc**

[image]

DanielTongAwesome, 2018-12-03:
I can send you my .sym file if you needed, I found this bug when I include double datatype inside the .sym

altendky, 2018-12-03:
@DanielTongAwesome, hopefully #268 fixes this for you.  There are all sorts of warnings about not running pip with sudo or as root etc...  but if you are running pip you can `pip install git+https://github.com/altendky/canmatrix@267-altendky-sym_double` to try out the PR and see if it works for you.

It does also look like you are running a fairly old version of canmatrix based on the error being on line 335 whereas it's now on 400-something.

DanielTongAwesome, 2018-12-03:
[DEMO_1_DOUBLE_NUCLEO_TX_STDID]
ID=400h
DLC=8
Var=Double double 0,64 /p:10

[DEMO_1_DOUBLE_NUCLEO_RX_STDID]
ID=401h
DLC=8
Var=Double double 0,64 /p:10

DanielTongAwesome, 2018-12-03:
Here is the example that I am using, please include in your .sym file and test and run the canconvert you will see the error message, seems other functions are good just this has some problems

altendky, 2018-12-04:
@DanielTongAwesome, thanks.  Were you able to try my fix?  It is merged into development now.

DanielTongAwesome, 2018-12-04:
Got it i will try it recently ~ @altendky  thanks for your help

closed 2019-02-25 by DanielTongAwesome