ebroecker/canmatrix #271: Question about arxml file can not be supported

BrushLee90, 2018-12-12:
When I run the test.py, it seems like that there are some "ImportError" errors during importing arxml, kcd, modules in formats.py.  The logs after running test.py shows as below.
Could you please help me fine what the problem is.
I'm looking forward for your reply.  Thank you sincerely in advance.
[image]
[image]

ebroecker, 2018-12-12:
did you
`pip install git+https://github.com/ebroecker/canmatrix#egg=canmatrix[arxml]`
or at least
`pip install -r requirements.txt`

BrushLee90, 2018-12-14:
Sorry,it's my mistake, I forgot it. 
It's OK now after installing.
Thank you very much for your attention in time !

closed 2018-12-14 by ebroecker