ebroecker/canmatrix #272: Error using cancompare

christianheussy, 2018-12-13:
Error when comparing two dbc files. Attempted using both python 3.4 and 3.7.

```python
File "c:\dbc_python\venv\lib\site-packages\canmatrix\compare.py", line 450, in dumpResult
    if type(res._changes[0]) == types.UnicodeType:
AttributeError: module 'types' has no attribute 'UnicodeType'
'''

altendky, 2018-12-13:
@christianheussy, we actually made it worse since whatever version you are running and we don't even `import types` anymore.  :]  But, #273 should have a fix/patch-over for this.  Let me know if it works for you.  You should be able to install such as:

```
yourenv\scripts\pip install git+https://github.com/altendky/canmatrix@272-altendky-unicode_type_check#egg=canmatrix
```

altendky, 2018-12-14:
@christianheussy, sorry...  normally my branches would be in my repository but this one ended up in the official repository.  The command should be:

```
yourenv\scripts\pip install git+https://github.com/ebroecker/canmatrix@272-altendky-unicode_type_check#egg=canmatrix
```

(this time I actually checked the url myself...)

christianheussy, 2018-12-14:
@altendky, that worked. I was able to install the patch and test cancompare successfully on several sets of dbc files.

altendky, 2018-12-14:
@christianheussy we'll wait for a review but it will probably be in the main `development` branch within a day or two most likely.

closed 2018-12-15 by altendky