ebroecker/canmatrix #275: Export value range(min/max) of SG signals to JSON

nodermatt, 2019-01-14:
First of all, thanks for the fantastic tool @ebroecker ! It was extremely helpful for us when we developed an App that needs to handle many different dbc files for various vehicle types.

I've got one question regarding the json extraction option (--jsonExportAll ): Is it possible to extract the valid value range of a CAN message from dbc to JSON also?

We noticed that when converting the below CAN message from dbc we're missing the min and max value defined in square brackets between the "unit" and "factor/offset". There is an issue [#126](https://github.com/ebroecker/canmatrix/issues/126 )about min/max/default but I wasn't sure if it is related to my question.

```
BO_ 1568 C2C_BCM_TimeSync: 6 C2C
 SG_ C2C_BCM_noUnixTimestamp : 0|40@1+ (1,0) [0|1099511627775] "s"  BCM
 SG_ C2C_BCM_UtcOffsetTimezone : 40|7@1+ (15,-720) [-720|840] "minutes"  BCM
 SG_ C2C_BCM_DST : 47|1@1+ (1,0) [0|1] ""  BCM
```

I thought I might be able to extend the code in cmjson.py [here] https://github.com/ebroecker/canmatrix/blob/b11574484ce03723a327de76f6b7a35c2aeaa250/src/canmatrix/cmjson.py#L74) as there is a min and max field for the "SG_" decoded signals which I found in dbc.py https://github.com/ebroecker/canmatrix/blob/b11574484ce03723a327de76f6b7a35c2aeaa250/src/canmatrix/dbc.py#L486

Is there a particular reason why the min and max fields aren't exported by default?

```
{
      },
      "comment": "",
      "id": 1568,
      "is_extended_frame": false,
      "length": 6,
      "name": "C2C_BCM_TimeSync",
      "signals": [
          "bit_length": 40,
          "comment": "UTC Zeit als Unix Zeitstempel (Sekunden seit 1.1.1970)",
          "factor": "1",
          "is_big_endian": false,
          "is_float": false,
          "is_signed": false,
          "name": "C2C_BCM_noUnixTimestamp",
          "offset": "0",
          "start_bit": 0,
          "unit": "s",
          "values": {}
        },
          "bit_length": 7,
          "comment": "Offset der Lokalen Zeitzone zu UTC in Minuten (ohne Sommerzeit)",
          "factor": "15",
          "is_big_endian": false,
          "is_float": false,
          "is_signed": false,
          "name": "C2C_BCM_UtcOffsetTimezone",
          "offset": "-720",
          "start_bit": 40,
          "unit": "minutes",
          "values": {}
        },
          "bit_length": 1,
          "comment": "Sommerzeit (+60 zu C2C_BCM_UtcOffsetTimezone um auf lokale Zeit zu kommen)",
          "factor": "1",
          "is_big_endian": false,
          "is_float": false,
          "is_signed": false,
          "name": "C2C_BCM_DST",
          "offset": "0",
          "start_bit": 47,
          "values": {}
        }
      ]
    }
```

ebroecker, 2019-01-15:
this is related to #136 - Json export is not yet complete.
No special reason.
I'll add a patch for it.

nodermatt, 2019-01-15:
Thanks for your prompt response +1

Was my analysis correct that the  "min" and "max" fields have to be added to the signalDict object in cmjson.py or are more changes necessary to achieve the goal?

ebroecker, 2019-01-15:
@nodermatt  please try my pr:

`yourenv\scripts\pip install git+https://github.com/ebroecker/canmatrix@iss275#egg=canmatrix
`

And yes, extending the signal-array would be enough.
I additionally added read min/max from json functionality and added tests for both reading and writing.

You can use this pr. It will go upstream. Typically it will take some days to do so.

Best Regards,
Eduard

closed 2019-02-15 by ebroecker