ebroecker/canmatrix #277: Facing issue in converting arxml to dbc

Jayanthdj, 2019-01-17:
[image]
 Could you please help

Jayanthdj, 2019-01-17:
If I use the current development branch then I am getting this below error.

DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 460 frames in arxml...
DEBUG - arxml - DEBUG 460 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 13933 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
DEBUG - arxml - Busname: WU_HCP4_CANFD
DEBUG - arxml - 71 frames found in arxml

DEBUG - arxml - processing Frame: FT_DEV_TME_Resp_00
DEBUG - arxml - CDD_DEV_TME_Resp_00_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU CDD_DEV_TME_Resp_00_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_DEV_TME_Resp_FF
DEBUG - arxml - CDD_DEV_TME_Resp_FF_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU CDD_DEV_TME_Resp_FF_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_BCM1_Req
DEBUG - arxml - ISO_BCM1_Req_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_BCM1_Req_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_BCM1_Resp
DEBUG - arxml - ISO_BCM1_Resp_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_BCM1_Resp_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_Funktionaler_Req_All
DEBUG - arxml - ISO_Funktionaler_Req_All_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_Funktionaler_Req_All_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_STH_Req
DEBUG - arxml - ISO_STH_Req_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_STH_Req_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_STH_Resp
DEBUG - arxml - ISO_STH_Resp_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_STH_Resp_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_Sitzmemo_BFS_Req
DEBUG - arxml - ISO_Sitzmemo_BFS_Req_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_Sitzmemo_BFS_Req_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_Sitzmemo_BFS_Resp
DEBUG - arxml - ISO_Sitzmemo_BFS_Resp_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_Sitzmemo_BFS_Resp_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_Sitzmemo_FS_Req
DEBUG - arxml - ISO_Sitzmemo_FS_Req_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_Sitzmemo_FS_Req_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_Sitzmemo_FS_Resp
DEBUG - arxml - ISO_Sitzmemo_FS_Resp_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_Sitzmemo_FS_Resp_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_TME_Req
DEBUG - arxml - ISO_TME_Req_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_TME_Req_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_TME_Resp
DEBUG - arxml - ISO_TME_Resp_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_TME_Resp_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_TSG_HBFS_Req
DEBUG - arxml - ISO_TSG_HBFS_Req_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_TSG_HBFS_Req_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_TSG_HBFS_Resp
DEBUG - arxml - ISO_TSG_HBFS_Resp_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_TSG_HBFS_Resp_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_TSG_HFS_Req
DEBUG - arxml - ISO_TSG_HFS_Req_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_TSG_HFS_Req_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_TSG_HFS_Resp
DEBUG - arxml - ISO_TSG_HFS_Resp_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_TSG_HFS_Resp_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_VSG_Req
DEBUG - arxml - ISO_VSG_Req_XIX_WU_HCP4_CANFD
DEBUG - arxml - DEBUG: AR4.x PDU ISO_VSG_Req_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_ISO_VSG_Resp
[... 148 lines cut ...]
DEBUG - arxml - DEBUG: AR4.x PDU ISOx_VSG_Resp_FD_XIX_WU_HCP4_CANFD no SIGNAL-TO-PDU-MAPPINGS found - no signal extraction!
DEBUG - arxml - processing Frame: FT_KN_BCM1
DEBUG - arxml - KN_BCM1_XIX_WU_HCP4_CANFD
DEBUG - arxml - No Compmethod found!! - try alternate scheme 1.
DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.
Traceback (most recent call last):
  File "C:\LegacyApp\Python27\Scripts\canconvert-script.py", line 9, in <module>
    load_entry_point('canmatrix==0+unknown', 'console_scripts', 'canconvert')()
  File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\convert.py", line 402, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
[... cut ...]
  File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 1379, in getFrame
    getSignals(pdusigmapping, newFrame, xmlRoot, ns, None, float_factory)
  File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 1223, in getSignals
    initvalue.text = guess_value(initvalue.text)
  File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 1215, in guess_value
    textValue = textValue.casefold()
AttributeError: 'str' object has no attribute 'casefold'

ebroecker, 2019-01-17:
Thanks,

you found a py2 incompatibility.
I'll provide a fix for this issue.
Meanwhile you could try using python3.3+

Jayanthdj, 2019-01-22:
Ok, Could you please give a fix for this issue with Python 2.7?

ebroecker, 2019-01-22:
please try branch
https://github.com/ebroecker/canmatrix/tree/iss277

`pip install git+https://github.com/ebroecker/canmatrix@iss277#egg=canmatrix `

Jayanthdj, 2019-01-24:
I got the same error with this patch :(  Could you please help as its quiet urgent

DEBUG - arxml - No Compmethod found!! - fuzzy search in syssignal.                                                                    DEBUG - arxml - processing Frame: FT_SMLWS_01                                                                                         Traceback (most recent call last):                                                                                                      File "C:\LegacyApp\Python27\Scripts\canconvert-script.py", line 9, in <module>                                                          load_entry_point('canmatrix==0+unknown', 'console_scripts', 'canconvert')()                                                         File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\convert.py", line 402, in main                    convert(infile, outfileName, **cmdlineOptions.__dict__)                                                                             File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\convert.py", line 43, in convert                  dbs = canmatrix.formats.loadp(infile, **options)                                                                                    File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\formats.py", line 64, in loadp                    return load(fileObject, importType, key, flatImport, **options)                                                                     File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\formats.py", line 74, in load                     dbs = moduleInstance.load(fileObject, **options)                                                                                    File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 1610, in load                     db.addFrame(getFrame(frameTrig,searchPoint,multiplexTranslation,ns, float_factory))                                                 File "C:\LegacyApp\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 1233, in getFrame                 idNum = int(idele.text)                                                                                                           AttributeError: 'NoneType' object has no attribute 'text'

ebroecker, 2019-01-24:
We help if we can, but please keep in mind, this is a spare time project!

This is not the same error, seems your ARXML does not provide an Arbitration-ID where canmatrix expects it.

If you want faster support, please provide the full debug-output or even better, provide your ARXML (maybe as private email).
Without we can only debug step by step and this will take its time.

Thanks

Jayanthdj, 2019-01-28:
Hi Edward, Sure. I have sent the arxml file to your mail. Please check it.

ebroecker, 2019-01-28:
Thanks, I got it - gonna check whats the difference in this "AR4.3" file

ebroecker, 2019-01-28:
added some "ignore-code" in  bb4faa98d7d6.
works at my side but only tested it with py3.7.

Jayanthdj, 2019-01-30:
Hi, Edward, I am sorry, It is not working on my side again. Could you please send me the converted .dbc and excel file please.

ebroecker, 2019-02-09:
at my side it works now and is already merged to development

closed 2019-02-09 by ebroecker

shubhamthakare, 2019-05-02:
Hi Jayanth

I want to convert an .arxml file to dbc .Please help me with it