ebroecker/canmatrix #281: Getting 0 frames in arxml with various .arxml files

andraflandra, 2019-02-05:
Hi, I am using an internal .arxml file I cannot show you but also a public one available here:
https://raw.githubusercontent.com/nikidimitrow/Learning-AUTOSAR-fundamental/master/BasicsOfAUTOSAR/MyECU.ecuc.arxml

I get this readout in cmd (win10):
`
C:\Python27\Scripts>canconvert -vv D:\user\ECUexample.arxml D:\user\ECUexampleAAA.xlsx
INFO - convert - Importing D:\user\ECUexample.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 0 frames in arxml...
DEBUG - arxml - DEBUG 0 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 0 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting D:\user\ECUexampleAAA.xlsx ...
INFO - convert - done
`

ebroecker, 2019-02-05:
Hi @andraflandra 
this is an `ECUC`-ARXML.

Conversion will only work with an `SYSTEM-DESCRIPTION`-ARXML

andraflandra, 2019-02-07:
How might I get this to work with ECUC-ARXML? Could your code be modified slightly or is a completely new solution needed?

ebroecker, 2019-02-08:
please try branch
https://github.com/ebroecker/canmatrix/tree/iss281

`pip install git+https://github.com/ebroecker/canmatrix@iss281#egg=canmatrix `

andraflandra, 2019-02-11:
First of all, thank you very much for your effort!

I can now convert the files but from the 1.329KB file all I get is a 6KB .xlsx table with only the first line of data as far as I can tell.

This is the readout in cmd:
INFO - arxml - seems to be a ECUC arxml. Very limited support for extracting canmatrix.
DEBUG - xls - DEBUG: Length of db.frames is 2

(this is the .arxml file I linked in the issue converted to .xlsx)
[ECUexampleCONVERTED.xlsx](https://github.com/ebroecker/canmatrix/files/2850639/ECUexampleCONVERTED.xlsx)

Also, the actual, non-example .arxml files I need to convert are still getting the line of "DEBUG 0 frames".
If there is any more info I can give you to make this work please let me know!,

Thanks again.

ebroecker, 2019-02-11:
@andraflandra 
what is the output of 
`canconvert -vv your_non_example.arxml  some_test.dbc`

Is there actual a Can-Bus defined in the ARXML?

andraflandra, 2019-02-11:
This is the output of said command in cmd:

```
C:\Users\user1>canconvert -vv D:\file.arxml D:\some_test.dbc
INFO - convert - Importing D:\file.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - DEBUG 0 frames in arxml...
DEBUG - arxml - DEBUG 0 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 0 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting D:\some_test.dbc ...
INFO - convert - done
```
However, this does not actually output/create any .dbc file. 
(but doing the line on the example code gets me the same 2 frame file but in .dbc)

As for your question, these are the opening lines of the file:
<?xml version="1.0" encoding="utf-8"?>

<AUTOSAR xsi:schemaLocation="http://autosar.org/schema/r4.0 AUTOSAR_4-2-1.xsd" xmlns="http://autosar.org/schema/r4.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">

ebroecker, 2019-02-11:
Hi @andraflandra ,

Thanks.
what kind of ARXML is this? In what context did you get this file?

The debug output looks like there is no CAN communication defined in this ARXML:
```
DEBUG - arxml - DEBUG 0 frames in arxml...
DEBUG - arxml - DEBUG 0 can-frame-triggering in arxml...
DEBUG - arxml - DEBUG 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - DEBUG 0 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
```

andraflandra, 2019-02-13:
It's from a model input, basically a database of host architecture for automotive. I do have a separate .dbc file that has CAN signals database on them. 

I'm new to this so please excuse my ignorance.

ebroecker, 2019-02-13:
If the ARXMLs describe a SWC (Software component) it will mostely certain not be any CAN information in it.

andraflandra, 2019-02-13:
There are SWC components included so I would assume you are correct. Does this make it impossible to use canmatrix to port arxml to xlsx or can it be enhanced perhaps?

ebroecker, 2019-02-13:
Sorry - this is a really complete other use case.
Something like extracting sender/receiver ports or/and exporting client/server ports from SWC-ARXML is a complete other work than exporting CAN communication information

andraflandra, 2019-02-14:
Thank you for all the help in any case. I will try to find a different solution or build my own.

closed 2019-02-23 by ebroecker