ebroecker/canmatrix #283: [No issue] release 0.7

danielhrisca, 2019-02-11:
Just wanted to thank you for releasing this version!

VincieD, 2019-02-11:
Hello to everyone,
do you plan on extending the canmatrix also for flexray and ethernet? If yes, how could i contribute to that?

BR, Vaclav

ebroecker, 2019-02-11:
@VincieD 
For both cases example data would be needed...

VincieD, 2019-02-11:
@ebroecker 
That should not be a problem. I could provide some arxmls examples. The difficult one will be for sure the ethernet, because there are not Frames available, but services and events. I can give a try to find some examples in next comming week.

altendky, 2019-02-11:
@VincieD are you talking about EtherCAN?  Or...  just regular old ethernet that you would push TCP etc over?

VincieD, 2019-02-11:
@altendky I'm talking about SOME/IP wich uses the TCP standard, it is just service oriented for automotive purposes
[http://some-ip.com/](url)

altendky, 2019-02-11:
@VincieD it'll be up to @ebroecker of course but after a _really_ quick glance it seems nearly entirely unrelated.  It's not another format for describing 11/29-bit identifiers and 0-8 byte payloads.  Nor an extension of that (like I think flexray is, or CANFD).  A library to support it might end up using canmatrix for the CAN aspects maybe?  But again, that was a _really_ quick glance so I might be way off...

friendlyranxiao, 2019-02-12:
hey, how to convert dbc file into a xlsx file?
and, how to convert the xlsx file into a dbc file?
so is there a tutorial to explain how to achieve that？

friendlyranxiao, 2019-02-12:
don't bother, I successfully convert the dbc into xlsx, and this is all most magic!

ebroecker, 2019-02-12:
@VincieD 
SOME/IP could make sense in my mind. 
There will be gateways from CAN to SOME/IP in future cars. 
I also can imagine that creating ARXMLs for SOME/IP is quite similar to creating CAN ARXMLs.
But I don't have much experience with SOME/IP...

For FLEXRAY its nearly the same. I just don't have to do much with FLEXRAY anyway. The only formats for defining a FLEXRAY network I know are fibex and ARXML. 

If you have a use case and maybe examples we could discuss if SOME/IP or FLEXRAY make sense in canmatrix or not. While SOME/IP is more future and FLEXRAY is more history in my mind...

VincieD, 2019-02-18:
@ebroecker 
SOME/IP is already in many cars. And i agree that Flexray will be not often used in future. But it is for instance for LiDAR, which means at least for next 2-3 years. 

I could provide arxml with SOME/IP specification if you would want to. Just let me know. And thank you for your time and support 👍

ebroecker, 2019-11-07:
I added some support in branch `arxml_sameip`

r142431, 2020-05-29:
@ebroecker @VincieD Hello, I would like to help you to improve this tool. What do you have in mind about this new feature? I didn't see anything about someip on that branch.

ebroecker, 2020-05-30:
Hi @r142431 

there is some very limited support for ethernet in this branch:

https://github.com/ebroecker/canmatrix/blob/f7e75cacb1de97b15bfda053dd495cf4e04a77c2/src/canmatrix/formats/arxml.py#L1615

r142431, 2020-06-09:
Hi @ebroecker, I was watching the code. My question is, what kind of features do you have in mind?. Would you like something like a some/ip database generator? or just want to take all of the attributes from the arxml

ebroecker, 2020-06-25:
@r142431 
I'm not sure what use cases are relevant.
a some/ip database generator sounds nice, but does someone need it?
How should it look like?

ebroecker, 2020-07-19:
hi @r142431 

I made some small improvements with some/ip.
Now it's possible to convert arxml to xlsx:
`convert.py --arxmlEthernet some_arxml_file.arxml some_excel_file.xlsx`

closed 2021-05-07 by ebroecker