ebroecker/canmatrix #284: .sym -> .dbc results in syntax error from CanDB++ (BA_DEF_ BO_ ...)

altendky, 2019-02-12:
@ebroecker, I'm hoping you know DBC well enough to have an idea for this.  I'll see if i can get CANdb++ (if it's free?) and try to see how it outputs such things.

```bash
altendky@lt:~$ venv/bin/pip freeze | grep canmatrix
canmatrix==0.7
```
```bash
altendky@lt:~$ venv/bin/canconvert Downloads/EPC_DG_ID247_FACTORY_tm3.sym x.dbc
```

(extensions change so github will accept it...)
[EPC_DG_ID247_FACTORY_tm3.sym.txt](https://github.com/ebroecker/canmatrix/files/2857046/EPC_DG_ID247_FACTORY_tm3.sym.txt)
[x.dbc.txt](https://github.com/ebroecker/canmatrix/files/2857014/x.dbc.txt)

[image]
CANdb++ version: 3.0.163

Line 6901:
```
BA_DEF_ BO_ "GenMsgCycleTime" INT 0 65535;
```

ebroecker, 2019-02-12:
@altendky 
usually you have to add 1 to the line candb++ claims.
I guess the defines for Bool are not correct. I'll double check this and give feedback...

altendky, 2019-02-12:
Maybe `BOOL` shouldn't have limits?  `STR` doesn't.  Is CANdb++ free?

ebroecker, 2019-02-12:
candb++ is not free as far as I know.

following patch creates dbcs which can be read from candb++ again...

```diff
diff --git a/src/canmatrix/sym.py b/src/canmatrix/sym.py
index 6e780e6..053c60c 100644
--- a/src/canmatrix/sym.py
+++ b/src/canmatrix/sym.py
@@ -330,12 +330,12 @@ def load(f, **options):
 
     db = CanMatrix()
     db.addFrameDefines("GenMsgCycleTime", 'INT 0 65535')
-    db.addFrameDefines("Receivable", 'BOOL False True')
-    db.addFrameDefines("Sendable", 'BOOL False True')
+    db.addFrameDefines("Receivable", 'STRING')
+    db.addFrameDefines("Sendable", 'STRING')
     db.addSignalDefines("GenSigStartValue", 'FLOAT -3.4E+038 3.4E+038')
-    db.addSignalDefines("HexadecimalOutput", 'BOOL False True')
+    db.addSignalDefines("HexadecimalOutput", 'STRING')
     db.addSignalDefines("DisplayDecimalPlaces", 'INT 0 65535')
-    db.addSignalDefines("LongName", 'STR')
+    db.addSignalDefines("LongName", 'STRING')
```

candb++ only allows following types:
[image]

altendky, 2019-02-12:
Works here as well, thanks.  I'll try to get to a PR soon.

ebroecker, 2019-02-12:
I'm not sure, maybe dbc exporter should at least warn if attribute types are not supported.
Or maybe dbc should replace not supported type by STRING...

altendky, 2019-02-12:
Errors sound good to me.  Either we have a bug internally or someone else has one in their code feeding the dbc export.  Though, maybe we should make what were bools into integers?  Or enumeration?

closed 2019-02-19 by ebroecker