ebroecker/canmatrix #288: put value table of a signal in just one cell

friendlyranxiao, 2019-02-15:
hello, Mr.

it is possible to put the value table of a signal in one cell rather than cells for each value?

hope the illustration may describe better:
[image]

friendlyranxiao, 2019-02-15:
each signal in each line, that looks neat, 
and user friendly for some one to find a signal in excel

ebroecker, 2019-08-19:
@friendlyranxiao 
this leads to a format which is not machine readabel. 
I'll add an export option for this.

ebroecker, 2019-08-19:
@friendlyranxiao 
could you try branch iss288 if it fullfills your needs?
there is a new commandline option : `--xlsValuesInSeperateLines` for this behaviour

ebroecker, 2019-08-20:
merged to development.
@friendlyranxiao please reopen, if its not what you wanted ....

closed 2019-08-20 by ebroecker