ebroecker/canmatrix #291: Release frequency

altendky, 2019-02-15:
What do you think about more frequent releases?  I'm trying to get away from vcs+commit-id dependency specification as it really isn't a great way to work.  Whether there's a theoretically issue may be a matter of opinion but pragmatically it's annoying.

Related topics that may be of interest:
1) Automatic release on git-tag + ci-success
2) [calver](https://calver.org/)

I just added [boots.py](https://github.com/altendky/boots) to [stlib](https://github.com/altendky/stlib).  It's a step along my dependency locking journey as well as my build-the-env-for-people effort.  Also has build and publish commands, even if they are pretty small amounts of code.

ebroecker, 2019-02-16:
YES more frequent releases would be very good!

I'm not fixed to one special system for versions. 
If you want to, just setup a new versioning system.

Funth0mas, 2019-03-03:
I'd first finish the typing annotation because it has already found some bugs. The `canmatrix` module is, say, already annotated. Now I'll do it with the formats, I hope it will be fast.

ebroecker, 2019-11-08:
... added auto deploy, so I do more often releases now.
currently I have in mind to release a 1.0 as next.
Afterwards maybe I switch to calver

timtianyang, 2019-11-13:
nice! what's the release cadence? Is it gonna be every week or month? Regardless I think some of the bug fixes are worth getting into pip sooner than later.

ebroecker, 2020-09-20:
auto-"magic" releasing is in ci, but due to lack of time I don't release that often.

Anyway I'll close this issue for now

closed 2020-09-20 by ebroecker