ebroecker/canmatrix #317: IndexError Occurs when parsing arxml file, involving tree.xpath()

BrushLee90, 2019-03-04:
After upgrading to the release 0.7 version, there is an index error occurring as below： 
[image]
The using test arxml file is listed:

[test_GEE.zip](https://github.com/ebroecker/canmatrix/files/2924837/test_GEE.zip)

BrushLee90, 2019-03-04:
In case you can not see the error snipping picture, I copy the error text as following:
Exception has occurred: IndexError
list index out of range
  File "D:\cantestpython\d\canmatrix\arxml.py", line 897, in getArPath
    baseElement = tree.xpath(xbasePath, namespaces=namespaceMap)[0]
  File "D:\cantestpython\d\canmatrix\arxml.py", line 928, in arGetChild
    ret = getArPath(xmlRoot, ret.text, namespace)
  File "D:\cantestpython\d\canmatrix\arxml.py", line 1033, in getSignals
    baseType = arGetChild(isignal,"BASE-TYPE", xmlRoot, ns)
  File "D:\cantestpython\d\canmatrix\arxml.py", line 1369, in getFrame
    getSignals(pdusigmapping, newFrame, xmlRoot, ns, None, float_factory)
  File "D:\cantestpython\d\canmatrix\arxml.py", line 1612, in load
    frameObject = getFrame(frameTrig,searchPoint,multiplexTranslation,ns, float_factory)
  File "D:\cantestpython\d\canmatrix\formats.py", line 74, in load
    dbs = moduleInstance.load(fileObject, **options)
  File "D:\cantestpython\d\canmatrix\formats.py", line 67, in loadp
    return None
  File "D:\cantestpython\d\temprun.py", line 11, in <module>
    arxmlobj = canmatrix.formats.loadp("test_GEE.arxml")

ebroecker, 2019-03-06:
@BrushLee90
Thanks for your bug report.
please give me a few days to analyze and fix.
Did you try 
```python
arxmlobj = canmatrix.formats.loadp("test_GEE.arxml", arxmlUseXpath = False) 
```
?

BrushLee90, 2019-03-08:
ok, it works when I add the parameter !  thanks a lot~

closed 2019-03-08 by BrushLee90